blackmagic-debug/blackmagic #1346: Found a possible security concern

benharvie, 2022-12-28:
Hello 👋

I run a security community that finds and fixes vulnerabilities in OSS. A researcher (@iamnynan) has found a potential issue, which I would be eager to share with you.

Could you add a `SECURITY.md` file with an e-mail address for me to send further details to? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) a security policy to ensure issues are responsibly disclosed, and it would help direct researchers in the future.

Looking forward to hearing from you 👍

(cc @huntr-helper)

dragonmux, 2022-12-28:
We will have a talk with Esden today and see what company address he'd like used. We'd be very interested to see what your researcher's found and get it fixed asap :)

esden, 2022-12-28:
I have just pushed a security policy including the email address where to report potential issues. Let us know if you have any more questions. :)

dragonmux, 2023-02-03:
This was dealt with in private and a fixed merged, so we're going to close this issue as fixed.

closed 2023-02-03 by dragonmux