blackmagic-debug/blackmagic #1353: Can't attach to Access Port on nRF52

giobauermeister, 2023-01-05:
I'm doing some tests with v1.9.0-rc1 and found that I can't attach to Access Port on nRF52.

I tested v1.8.2 https://github.com/blackmagic-debug/blackmagic/commit/5cd430647ecfb6dab4a2ed858fb98567723de699 and v1.9.0-rc0 https://github.com/blackmagic-debug/blackmagic/commit/14e9df69fa44f44a4f3906db581d1289c2329bc6. They seem to be working fine.

```
(gdb) target extended-remote /dev/ttyBmpGdb
Remote debugging using /dev/ttyBmpGdb
(gdb) monitor swdp_scan
Target voltage: 3.18V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 2
Attaching to Remote target
Ignoring packet error, continuing...
```

zyp, 2023-01-05:
I can reproduce this with BMDA:

```
(gdb) target ext :2000
Remote debugging using :2000
(gdb) mon swdp_scan
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) attach 2
Attaching to Remote target
Remote connection closed
```

BMDA is segfaulting:
```
Black Magic Debug App v1.9.0-rc1
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 1209:3443 2FCC8092A2189006 Orbcode
 Orbtrace
Listening on TCP port: 2000
scan_multidrop: false
Segmentation fault: 11
```

dragonmux, 2023-01-05:
Ack, we'll check the other special AP functions (this isn't the only one) and verify they set the target structure up properly. We're surprised this worked before as we replaced the tdesc stuff like-for-like - we'll dive into a suitable way to fix this, even if that ends up being an if guard in target_regs_description:

https://github.com/blackmagic-debug/blackmagic/blob/98915144944d46cb2dc1daedc734b0134dbde7fb/src/target/target.c#L471-L474

zyp, 2023-01-05:
> We're surprised this worked before as we replaced the tdesc stuff like-for-like…

It had an if-guard in the form of a ternary before:

```c
return t->tdesc ? t->tdesc : "";
```

dragonmux, 2023-01-05:
Right, but t->regs_description() can return NULL (for allocation failure) and we cannot return "" there as `free("")` == crash which is why the calling code should do NULL return translation now. that does suggest it should be an if statement guarding the call to regs_description() and a fall-back `return NULL;` though - but we need to check that results in suitable behaviour or if the correct fix is actually to set t->regs_description for those fallback APs all the same

zyp, 2023-01-05:
I figure that in any case returning an empty target description to gdb is reasonable since there's no CPU attached to that AP. There's then two natural solutions to this problem; either we populate `regs_description` with a dummy function that mallocs and returns an empty string (i.e. only the end terminator) or we allow it to be null, make `target_regs_description` return a null string pointer when `regs_description` is a null function pointer and check for null in `exec_q_feature_read`.

I think both are equally valid solutions, with a fairly even tradeoff (cost of malloc vs cost of extra checking), so I suggest going with whichever seems to have less complexity and maintenance overhead and is least likely to cause surprises in the future.

dragonmux, 2023-01-05:
This should now be fixed - please see the linked PR

zyp, 2023-01-06:
PR fixes the original issue, but now it's segfaulting at a different point instead, which doesn't seem to have been introduced with a recent change, but is likely platform dependent.

It's now crashing when GDB sends a `p` packet:
```c
		case 'p': { /* Read single register */
			ERROR_IF_NO_TARGET();
			uint32_t reg;
			sscanf(pbuf, "p%" SCNx32, &reg);
			uint8_t val[8];
			size_t s = target_reg_read(cur_target, reg, val, sizeof(val));
			if (s > 0)
				gdb_putpacket(hexify(pbuf, val, s), s * 2U);
			else
				gdb_putpacketz("EFF");
			break;
		}
```
It's crashing because `target_reg_read` returns garbage (4295094017 in my case), which causes a buffer overflow in `hexify`. This happens because `reg_read` defaults to `nop_function`, which is implemented like this:
```c
static bool nop_function()
{
	return true;
}
```
… and gets compiled to this:
```
blackmagic`nop_function:
    0x10001ef0a <+0>: pushq  %rbp
    0x10001ef0b <+1>: movq   %rsp, %rbp
    0x10001ef0e <+4>: movb   $0x1, %al
    0x10001ef10 <+6>: popq   %rbp
    0x10001ef11 <+7>: retq   
```
The issue is that since it's returning a bool, it's only setting `al`, i.e. the lower 8 bits of the full `rax` register, so when it's cast to fit into a function pointer returning `ssize_t`, the upper bits of the return value contains garbage.

The right solution in this case would be to use a different dummy function for `reg_read` (and possibly some of the other function pointers) that instead returns 0 as a `ssize_t`.

dragonmux, 2023-01-06:
That's just outright wrong and should be implemented using a nop function that returns 0.. good catch!

This highlights why we've been trying to turn more warnings on as this is actually something the compiler can warn about. At least it's another simple fix. We'll open another PR about it specifically

dragonmux, 2023-01-06:
Please give #1356 a whirl, zyp - we've done our best to properly replace and fix that undefined mess from before, which should result in smooth operation and good behaviour

giobauermeister, 2023-01-06:
I tested #1356 on top of #1355 and it seems to be working now.

giobauermeister, 2023-01-10:
> I tested #1356 on top of #1355 and it seems to be working now.

I was just testing latest commit eca88091e2c7489446c505224010b5c3129b7c33 and it seems that there is an error again trying to attach to Access port on nRF52. But now there are warnings about XML target description. Strange because testing #1356 on top of #1355 seemed to be working. Maybe I tested wrong?

```
(gdb) target extended-remote /dev/ttyBmpGdb
Remote debugging using /dev/ttyBmpGdb
(gdb) monitor swdp_scan
Target voltage: 3.19V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 2
Attaching to Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
Ignoring packet error, continuing...
```

dragonmux, 2023-01-11:
This represents expected behaviour - the AP does not represent a real core, and defines no target registers description function, resulting in the firmware correctly giving an empty reply for the request packet, and hence the warning.

Because of when the qSupported packet that triggers all that is sent, and our need to send the target registers description for things like Cortex-M*F cores to debug correctly, there is likely no fix for the warning.

giobauermeister, 2023-01-13:
@dragonmux Yes, now I understand the reasons for the warning. But my real point is that I was not able to attach to the Access Port.

dragonmux, 2023-01-13:
ah, ok; noted! A debug build of the firmware or using a `ENABLE_DEBUG=1` build BMDA to see what BMP is doing when it fails to attach would be very useful. Given that sounds like it's broken again, we'll reopen the issue

TheyCallMeNobody, 2023-02-19:
Please try my fix. The problem seemed to be, that it tried to read the Registers of the target, which aren't existing. Fixed this by implementing a dummy register Response

PR #1355 (2023-01-07): Fix: Target registers description crash

## Detailed description

This PR addresses the issue reported in https://github.com/blackmagic-debug/blackmagic/issues/1353 which causes a crash due to calling a NULL pointer in `target_regs_description`. This occurs because not all targets set the t->regs_description member. This fix restores the previous behaviour that was changed in 2c52af6 and improves robustness.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines [... cut]
closed 2023-02-21 by dragonmux