blackmagic-debug/blackmagic #1358: Support a la STLINK V3 Bridge API?

nmschulte, 2023-01-10:
Hi, I wonder if the firmware has any support for generic serial interfacing, like the STLINK V3 Bridge API enables?

Thanks!

https://www.st.com/en/development-tools/stlink-v3-bridge.html

https://blog.baneu.net/st-link-bridge-i2c-eeprom/

dragonmux, 2023-01-10:
We can't tell much from those links about how it works - what the host sees, etc - but based on the block diagrams, the short answer is no.

The slightly longer answer is that: no, the BMD firmware implements only a UART serial to USB conversion for reasons of hardware limitations in the majority of supported probes, and because we have a limited set of endpoints available which are (almost) all used up by supporting 2x USB serial interfaces (3 endpoints each), and Trace/SWO (another endpoint). In a future release we will support using the main JTAG/SWD interface for SPI as well, allowing access to SPI Flash attached via the unified connector, but it was not possible to add this as a second interface along side the main JTAG/SWD one in the native hardware.

If such functionality was ever added, it would be extremely limited in scope and as it would not benefit native hardware, it is unlikely we'd be the one writing the PR.

nmschulte, 2023-01-10:
I appreciate the fast response, thanks!  I just stumbled upon Black Magic and was interested to try the firmware on my STLINK V3.

I agree the information about the bridge API is lacking: apparently there's support in the ST firmware which corresponds with this API, enabling one to write code to interface with devices (presumably from the target device, but generically) via serial protocols (SPI, I2C, CAN, etc.).

dragonmux, 2023-01-10:
Well, it might not contain the bridge functionality, but there is a PR open for ST-Link v3 as a firmware platform - see https://github.com/blackmagic-debug/blackmagic/pulls/1354. We're aiming to have it merged for v1.10.