blackmagic-debug/blackmagic #1363: Fix BMP source compatability with the GCC 12 ARM toolchain

dragonmux, 2023-01-20:
Currently BMP is built and tested against the GCC 10 ARM toolchain for Cortex-R and Cortex-M parts. This is fine, except that toolchain is out of support and ARM's current toolchain is based on GCC 12. A lot changed in 11 and 12 and BMP needs updating to handle that. Additionally, this demands we sort two issues that are outstanding:

1) UB in the BMP codebase which is causing miscompilation and results in non-booting binaries
2) Incomplete handling of the newlib libc faked file IO layer and no implementation of the reentrant versions of any of the functions, resulting in mess like this from the compiler:

```
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-fstatr.o): in function `_fstat_r':
fstatr.c:(.text._fstat_r+0xe): warning: _fstat is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-signalr.o): in function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-isattyr.o): in function `_isatty_r':
isattyr.c:(.text._isatty_r+0xc): warning: _isatty is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-signalr.o): in function `_kill_r':
signalr.c:(.text._kill_r+0xe): warning: _kill is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x10): warning: _lseek is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x10): warning: _read is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
```

koendv, 2023-02-13:
Also:
blackmagic/src/target/nxpke04.c:301:53: error: cast increases required alignment of target type [-Werror=cast-align]

dragonmux, 2023-02-17:
So.. diving into this particular set of linker warnings reveals that they only happen on non-`ENABLE_DEBUG=1` builds and they are specifically caused by `-spec=nosys.spec`. The only fix is either to entirely eliminate all use of the newlib-nano functions that trigger the inclusion of any part of the printf() family and friends, or to provide stubs compiled in when `ENABLE_DEBUG` is falsey that are nop's for all of those

We have not yet been able to reproduce the error that Koen reported, however the exact line number should mean a dry-run can be accomplished to fix and address that issue too

dragonmux, 2023-02-24:
@koendv Please give the above linked PR a spin as it should fix your issue and the linker warnings.

koendv, 2023-02-24:
ok. fixes the error in nxpke04.c
Perhaps newlib stubs better in separate file; for those platforms that have newlib but do not use usb_serial.c.

dragonmux, 2023-02-24:
Not aware of any platform for which that is the case that's in-repo, and tbqh moving the stubs means needing to reorganise the printf()/_write() code at the bottom of usb_serial.c anyway - but you make a point that the organisation could do with some further tidy-up.

Thank you for confirming the NXP KE04 issue is fixed :)

closed 2023-04-02 by esden