blackmagic-debug/blackmagic #1365: Support for stm32f030f4p6

mirko, 2023-01-22:
Hello,

I'm having two cheap STM32F030F4P6 boards and an STM32F103 board.
BMP is flashed with latest firmware 1.9.0-rc1.
Connecting VCC-ref, GND, SWDIO and SWDCLK to the STM32F103 board and running `monitor swdp_scan` correctly detects the targets.
When connecting the two (one at a time, obviously) STM32F030 based board the same way, though, they are *not* detected.
All I see is `SW-DP scan failed!` when running `monitor swdp_scan`.
Is this to be expected? Is support missing for the STM32F030 series?
What could I be doing wrong? What could I do to eventually connect to those via SWD via my BMP?

dragonmux, 2023-01-22:
Most likely you're being affected by #1359 which has a fix for it merged on main

mirko, 2023-01-22:
Thanks, that sounded wonderfully easy.
Unfortunately I'm still having then very same issue with v1.8.2 :/

dragonmux, 2023-01-22:
Could we ask for you to try a debug build of main (strip out a few unused targets from src/Makefile to get the build to fit, build with `make PROBE_HOST=native ENABLE_DEBUG=1`) and get a trace from the debug serial port please? Run `monitor debug_bmp enable` before doing the scan and the debug build of the firmware will log to the secondary USB serial port what it's doing.

dragonmux, 2024-10-10:
We do believe this issue has been fixed though this is still pending testing - if in a couple of weeks we still have not had a response, we will close this as fixed in either v1.10 or on `main`.

closed 2024-11-09 by dragonmux