blackmagic-debug/blackmagic #1370: BMDA: RP2040 does not connect via dap42 (CMSIS-DAP) adapter

dmsc, 2023-01-29:
Trying to debug a RP2040 MCU with the dap42 debugger, via CMSIS-DAP interface, the SWD scan fails:
```
(gdb) tar ext :2000
Remote debugging using :2000
(gdb) mon s        
Exception: SWDP invalid ACK
SW-DP scan failed!
Failed
```

Problem is with target selection after a forced error, at the end of the [log file](https://github.com/blackmagic-debug/blackmagic/files/10530429/bad2.log):
```
 command: 06 00 04 00 0f 
response: 06 04 00 01 0d 00 00 00 10 00 00 00 05 00 00 00 b1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
dap_mem_read transferred 4 blocks
ap_memread @ e00ffff0 len 16: 0d 00 00 00 10 00 00 00 05 00 00 00 b1 00 00 00
dap_swd_dp_error (protocol recovery? false)
-> dap_swj_sequence (64 cycles)
 command: 12 40 ff ff ff ff ff ff ff 0f 
response: 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 command: 05 00 01 02 
response: 05 01 01 77 24 c1 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
        dap_read_reg 00 0bc12477
        dap_write_reg 0c 01002927
 command: 05 00 01 0c 27 29 00 01 
response: 05 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
dap_write_reg 0c data 01002927:fault
 command: 05 00 01 06 
response: 05 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

The code is at cmsis_dap.c, line 282:
```
static uint32_t dap_swd_dp_error(adiv5_debug_port_s *dp, const bool protocol_recovery)
{
	DEBUG_PROBE("dap_swd_dp_error (protocol recovery? %s)\n", protocol_recovery ? "true" : "false");
	if (dp->version >= 2 || protocol_recovery) {
		/* On protocol error target gets deselected.
		 * With DP Change, another target needs selection.
		 * => Reselect with right target! */
		dap_line_reset();
		dap_read_reg(dp, ADIV5_DP_DPIDR);
		dap_write_reg(dp, ADIV5_DP_TARGETSEL, dp->targetsel);
		/* Exception here is unexpected, so do not catch */
	}
```

The line with `dap_write_reg(dp, ADIV5_DP_TARGETSEL, dp->targetsel);` fails, as the probe returns error 7 = NO_ACK (no response from target). Note that from the reset all read and write commands fail.

Any ideas?

dmsc, 2023-01-29:
Additional information: If I comment out the line: `dap_write_reg(dp, ADIV5_DP_TARGETSEL, dp->targetsel);`, the scan works, but it drops on attach:

```
(gdb) tar ext :2000
Remote debugging using :2000
(gdb) mon s        
Available Targets:
No. Att Driver
 1      Raspberry RP2040 M0+
 2      Raspberry RP2040 M0+
 3      Raspberry RP2040 Rescue (Attach to reset!) 
(gdb) attach 1     
Attaching to program: prog.elf, Remote target
Attaching to Remote target failed
(gdb) 
```

Attached new log: [bad3.log](https://github.com/blackmagic-debug/blackmagic/files/10530467/bad3.log)

dragonmux, 2023-01-29:
We know what the bug is - and we forgot about this instance of the error handler when fixing the bug in the main firmware. the no-response response is actually correct, however the dap_write_reg() call for TARGETSEL must come before the DPIDR read for fun protocol reasons and then dp->fault needs resetting to 0 after the write.

This is another instance of https://github.com/blackmagic-debug/blackmagic/issues/1359 which needs a mix of e23c78 and ba5c56 applying to this situation to correctly fix

dragonmux, 2023-01-29:
The fix for this needs a second version of dap_write_reg(), however a partial fix is now available on branch fix/bmda-cmsis-dap-protocol-recovery. We will deal with the register write problem a bit later.

dmsc, 2023-01-29:
Thank for your reply. I tried the fix/bmda-cmsis-dap-protocol-recovery branch over the PR #1369, now the scan works but the attach still errors out:

[bad4.log](https://github.com/blackmagic-debug/blackmagic/files/10530618/bad4.log)

The write and the read both return error 7 (NO_ACK).

dragonmux, 2023-01-29:
We have updated the fix branch to address the lingering issue of dap_write_reg() being used in the error recovery path and impeding the ability to complete protocol recovery. Please let us know how you get on with the branch

dragonmux, 2023-01-29:
If this issue is not entirely solved, please let us know and we'll re-open it and work on further fixes for v1.10

dmsc, 2023-01-29:
Hi!

Sadly, it does not work yet, this is the current log: 
[bad6.log](https://github.com/blackmagic-debug/blackmagic/files/10530982/bad6.log)

As you can see, the scan works ok, but then at attach, the commands fail.

dragonmux, 2023-01-29:
That smacks of some of the deeper problems with the CMSIS-DAP implementation as it currently stands, which we're set to address in v1.10. We'll double-check that once we've had some sleep but for now we've re-opened the issue and re-marked it for v1.10.

Depending on what needs doing exactly to fix this (it's re-treading the same ground we fixed in the firmware caused by old code not dealing with protocol recovery properly and sending the logic into a loop when a protocol recovery event is triggered) we may well also backport the fix to v1.9 as a point release. We'll keep this issue updated with our progress.

dmsc, 2023-01-29:
Thank you!.

Let me know if I can help you, currently pyOCD and openocd work with dap42 and rp2040, but I think that the workflow with BMDA is a lot better.

zyp, 2023-01-31:
As discussed on Discord, I'm seeing what appears to be the same issue on nrf91 (which also got DPv2).

dmsc, 2023-01-31:
Hi!

Analyzing the log, the last DAP commands sent are:
```
cmd (12) SWJ_Sequence       : 40 ff ff ff ff ff ff ff 0f
rsp (12) SWJ_Sequence       :
cmd (05) Transfer           : 00 01 0c 27 29 00 01
rsp (05) Transfer           : 00 07
cmd (05) Transfer           : 00 01 02
rsp (05) Transfer           : 00 07
```
The SWJ sequence is the line reset, 60 ones and 4 zeroes, and then the TARGETSEL register is written with the target value.

But this is not OK, as the write should be done using the multi-drop target selection, with the mandatory not-driven cycles. This is implemented in the `dap_dp_low_write` function, using a SWD_Sequence command.

So, I applied the following patch:
```diff
diff --git a/src/platforms/hosted/cmsis_dap.c b/src/platforms/hosted/cmsis_dap.c
index 635281be..d1a3061f 100644
--- a/src/platforms/hosted/cmsis_dap.c
+++ b/src/platforms/hosted/cmsis_dap.c
@@ -87,6 +87,9 @@ typedef enum cmsis_type {
 typedef struct hid_device_info hid_device_info_s;
 #endif
 
+/*- Forward declaration -----------------------------------------------------*/
+static bool dap_dp_low_write(adiv5_debug_port_s *dp, uint16_t addr, const uint32_t data);
+
 /*- Variables ---------------------------------------------------------------*/
 static cmsis_type_e type;
 static libusb_device_handle *usb_handle = NULL;
@@ -292,7 +295,7 @@ static uint32_t dap_swd_dp_error(adiv5_debug_port_s *dp, const bool protocol_rec
                 */
                dap_line_reset();
                if (dp->version >= 2)
-                       dap_write_reg_no_check(dp, ADIV5_DP_TARGETSEL, dp->targetsel);
+                       dap_dp_low_write(dp, ADIV5_DP_TARGETSEL, dp->targetsel);
                dap_read_reg(dp, ADIV5_DP_DPIDR);
                /* Exception here is unexpected, so do not catch */
        }
```

With this change, the target selection works, but sadly the debug still does not work :-(

Attached is the log with the above patch applied: [bad7.log](https://github.com/blackmagic-debug/blackmagic/files/10542304/bad7.log)

dmsc, 2023-01-31:
Also,  @zyp , can you test with your nrf91?

Perhaps this change makes it work.

dmsc, 2023-01-31:
PR #1377 solves this issue, there were two different problems.

zyp, 2023-01-31:
I can confirm #1377 appears to solve the problem for me too.

PR #1371 (2023-01-29): Fix: BMDA's CMSIS-DAP protocol recovery

## Detailed description

Due to some code duplication in BMDA (which itself needs addressing but will be taken care of in v1.10), a second instance of #1359 was laying in wait in the CMSIS-DAP implementation. This PR addresses this second copy of the buggy protocol recovery code.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.m [... cut]
PR #1377 (2023-01-31): Fix CMSIS-DAP access to RP2040
## Detailed description

This fixes #1370:

- Fixes attaching to multi-drop targets by using correct write sequence.
- Fixes memory reading, allowing to correctly decode ROM tables.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this repository
* [x] It builds for hardware native (`make PROBE_HOST=native`)
* [x] It builds as BMDA  [... cut]
commit 10b1cc0e (2023-01-31): hosted/cmsis_dap: Fixes CMSIS-DAP multi-drop targets.

To select targets in SWD, we need to perform a special write sequence,
this is done using dap_dp_low_write instead of dap_write_reg.

As we don´t use dap_write_reg_no_check anymore, just remove the
function.

This partially fixes #1370.
closed 2023-01-31 by dragonmux