blackmagic-debug/blackmagic #1395: new target - Puya Py32x

prosper00, 2023-02-22:
I've been playing with a new mcu, the Puya py32f030, a cortex M0+ device, after reading about it on jaycarlson's site ( https://jaycarlson.net/2023/02/04/the-cheapest-flash-microcontroller-you-can-buy-is-actually-an-arm-cortex-m0/ ). I am able to connect to it using daplink, and programs like pyocd work with it when provided the CMSIS pack file. BMP (hosted, w/daplink) reports:

$ src/blackmagic -E
Black Magic Debug App 308aec7
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using c251:f001 LU_2022_8888 jixin.pro
 CMSIS-DAP_LU
CMSIS_DAP < v1.2 can not handle multi-drop, disabling
scan_multidrop: false
dap_read_single failed (fault = 4)
dap_read_single failed (fault = 4)
Please report unknown device with Designer 0x43b Part ID 0x4c0

Documentation (including .pack file) is available from the vendor ( https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU%E8%B5%84%E6%96%99-20221117.rar ). There's also a really handy gcc/make/pyocd build environment someone set up here ( https://github.com/IOsetting/py32f0-template )

dragonmux, 2023-02-22:
Could you please run `src/blackmagic -tv 1` and share the results as that will give us significantly more information to go on for how to identify the part and how far BMD is getting with the identification process

prosper00, 2023-02-23:
$ src/blackmagic -tv1
Black Magic Debug App 308aec7
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using c251:f001 LU_2022_8888 jixin.pro
 CMSIS-DAP_LU
Using hid transfer
Ver 1.0, Cap (0x 3): SWD/JTAG
Running in Test Mode
Target voltage: (null) Volt
Speed set to  4.0000 MHz for SWD
CMSIS_DAP < v1.2 can not handle multi-drop, disabling
scan_multidrop: false
DP DPIDR 0x0bc11477 (v1 MINDP rev0) designer 0x43b partno 0xbc
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=e00ff003 CSW=03000040 (AHB-AP var3 rev0)
Halt via DHCSR(01030003): success after 6ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c0
0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x00000004000bb008  DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
Calling lpc11xx_probe
dap_read_single failed (fault = 4)
dap_read_single failed (fault = 4)
Please report unknown device with Designer 0x43b Part ID 0x4c0
1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x00000004000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
ROM: Table END
***  1   Unknown ARM Cortex-M Designer 43b Part ID 4c0 M0+

closed 2024-05-12 by dragonmux