blackmagic-debug/blackmagic #1402: support debugging stm32u5

jerabaul29, 2023-02-24:
The stm32u5 is starting to be a really interesting family of MCUs: https://www.st.com/en/microcontrollers-microprocessors/stm32u5-series.html . Ultra low power, up to 4MB Flash and 2.5MB RAM, plenty of security features, 2D GPU, etc.

Any hope to get support for these MCUs? :) Is it something that I can typically expect to see in the future? :)

UweBonnes, 2023-02-27:
The U5 Nucleos carry an STLink -V3.  STLinkV3 supports most of the features from a BMP hardware probe by using the blackmagic application. Some low level JTAG/SWD things can not be mapped, but that is not needed to talk to the  U5. The special registers  and security feature of the  Cortex-M33 are not yet supported and that will need some deeper digging, but I think you can already start by using the  blackmagic application with that board.

UweBonnes, 2023-02-27:
I reads like [jerabaul29] is not able to buy a BMP hardware probe at the moment due to part shortage. With no probe he yhinks he can not work on U5 support. But Blackmagic application is perfect for the kind of work needed to get a new Cortex-M target running. No need for constant reflash, and debug messages appear on the starting console without the constant need to set up the debugging console on blackmagic hardware...

jerabaul29, 2023-02-27:
Thanks for the explanation.

I was not aware that the board has a stlink built in (I have on my todo list to look at it but not time so far), thanks for pointing this out :) .

But independently of this specific nucleo board having an stlink built-in, it could be great to support BMP debugging of the stm32u5 family natively for other boards that may not come with the stlink built in and may only expose the native stm32u5 chip debug pins :) .

perigoso, 2023-02-27:
@UweBonnes, I don't know where you got that:
1. they were not able to buy a BMP (they actually just said they were going to order one, and gave no feedback as to whether they were able to)
2. they imply they can't work on the target without one (they don't suggest they want to work on supporting it directly either, other than testing)

That said, it is likely they won't be able to get a BMP right now and BMDA is a good start if they want to work on it(even if it is generally fairly broken)

jerabaul29, 2023-02-27:
I am indeed struggling to find a BMP, I have put myself on notification lists a couple of places. And I was not aware of the built in debugger, I am a bit of a n00b. Thanks all for the help and pointers :) .

perigoso, 2023-02-27:
> But independently of this specific nucleo board having an stlink built-in, it could be great to support BMP debugging of the stm32u5 family natively for other boards that may not come with the stlink built in and may only expose the native stm32u5 chip debug pins :) .

@jerabaul29 adding support via BMDA would mean support via the BMP too, indirectly, the problem is that there is no perfect parity between the two, and it is likely that there are bugs that show up only when using the actual BMP (which is the real target of the application, BMDA is just a tool)

UweBonnes, 2023-02-27:
Blackmagic hardware will profit on any work and test done on target support with Blackmagic application.

perigoso, 2023-02-27:
> I am indeed struggling to find a BMP, I have put myself on notification lists a couple of places. And I was not aware of the built in debugger, I am a bit of a n00b. Thanks all for the help and pointers :) .

I believe they are out of stock everywhere right now, you might want to contact @esden, but you probably will have to wait for a new batch

jerabaul29, 2023-02-27:
I will wait, no worries - this is a side hobby project anyways, so my time for it comes and goes :) .

dragonmux, 2023-10-18:
Solved by #1574 which added support. Closing as fixed.

closed 2023-10-18 by dragonmux