blackmagic-debug/blackmagic #1403: making it clear what is open source vs non open source

jerabaul29, 2023-02-24:
If I understand well, this repository contains the full firmware of the blackmagic debug probe, is this correct? So, the firmware of the blackmagic debug probe is then open source if I understand well?

But I cannot find the hardware design / electronics cad files for the blackmagic debug probe here or another place on the internet. Does that well mean that the hardware design itself is not open source?

Could this be make clearer in the readme? I have been recommended to use this debug probe as it is open source, but if I understand well, actually, the firmware is open source, but the hardware is close source?

perigoso, 2023-02-24:
The hardware too is open-source but is available by request only, contact @esden 
The correct place for this information is https://black-magic.org/hardware.html, there could be a note about the contact only though I agree

jerabaul29, 2023-02-24:
Ok, thanks for the explanation and pointer :) .

Is it not a bit strange / unusual to have "open source hardware design not available openly on the project repository but available upon request"? Any reason for not just having the electronics cad file etc (like, a kicad project or similar, whatever tool is used) directly available on this repo in a specific folder? :)

For my part, I will be more willing to buy a blackmagic probe instead of a less expensive JLink Edu mini for my hobby projects if I have a feeling that the blackmagic probe is fully open source hardware and software :) .

jerabaul29, 2023-02-24:
(about ordering: of course that is when / if I manage to find a shop that is not sold out :) ).

dragonmux, 2023-02-24:
> Many thanks for taking the time to patiently answer my questions and point to the relevant places :) . That convinces me to go for buying a BMP myself :) .
> 
> My 2 cents on this: it may be good to write a short paragraph with a few lines about this on the readme maybe? I am used (and I think most people around me do the same) to scan readmes for a few lines about this kind of points when looking at a project and trying to judge if it does is open source or not :) .

Addressing this is on Esden's TODO list along with how best to release schematics or a block diagram of the v2.3b hardware, however other concerns of the business have been dominating his time the last 3 months as he tries to sort production for the Glasgow Crowd Supply campaign out. It will get addressed once he's got some more free time for that. Many people have enquired about the schematic for the board since v2.3b's release.

> 
> Fell free to close (or keep open until a few readme lines are added if you feel this sounds reasonable).

We'll keep the issue open but change the title so we have a tracking issue for fixing this