blackmagic-debug/blackmagic #1417: Error using make: arm-none-eabi-gcc: No such file or directory

lenvm, 2023-03-12:
I receive a number of `arm-none-eabi-gcc: No such file or directory` errors when trying to build the code.

Version information:
- **Computer**: Apple silicon M1
- **OS**: MacOS Ventura 13.2.1
- **arm-none-eabi-gcc version**: Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/arm-none-eabi-gcc: Mach-O 64-bit executable arm64
- **issue reproduced on blackmagic commit**: 32543bb (v1.9.0) and 61d3163 (latest main)

I did the following:
```
git clone git@github.com:blackmagic-debug/blackmagic.git
cd blackmagic
git submodule update --init --recursive
make PROBE_HOST=blackpillv2
```

The output is:
```
  GENHDR  stm32/f0
  GENHDR  stm32/f1
  GENHDR  stm32/f2
  GENHDR  stm32/f3
  GENHDR  stm32/f4
  GENHDR  stm32/f7
  GENHDR  stm32/l0
  GENHDR  stm32/l1
  GENHDR  stm32/l4
  GENHDR  stm32/g0
  GENHDR  stm32/g4
  GENHDR  stm32/h7
  GENHDR  gd32/f1x0
  GENHDR  lpc13xx
  GENHDR  lpc17xx
  GENHDR  lpc43xx/m4
  GENHDR  lpc43xx/m0
  GENHDR  lm3s
  GENHDR  msp432/e4
  GENHDR  efm32/tg
  GENHDR  efm32/g
  GENHDR  efm32/lg
  GENHDR  efm32/gg
  GENHDR  efm32/hg
  GENHDR  efm32/wg
  GENHDR  efm32/ezr32wg
  GENHDR  nrf/51
  GENHDR  nrf/52
  GENHDR  sam/3a
  GENHDR  sam/3n
  GENHDR  sam/3s
  GENHDR  sam/3u
  GENHDR  sam/3x
  GENHDR  sam/4l
  GENHDR  sam/d
  GENHDR  vf6xx
  GENHDR  swm050
  GENHDR  pac55xx
  BUILD   lib/stm32/f1
  CC      adc.c
make[2]: arm-none-eabi-gcc: No such file or directory
make[2]: *** [adc.o] Error 1
  BUILD   lib/stm32/f4
  CC      adc_common_v1.c
make[2]: arm-none-eabi-gcc: No such file or directory
make[2]: *** [adc_common_v1.o] Error 1
  BUILD   lib/lm4f
  CC      assert.c
make[2]: arm-none-eabi-gcc: No such file or directory
make[2]: *** [assert.o] Error 1
 GEN include/version.h
  CC      target/adiv5.c
make[1]: arm-none-eabi-gcc: No such file or directory
make[1]: *** [adiv5.o] Error 1
make: *** [all] Error 2
```

These paths are defined in [Makefile line 20](https://github.com/blackmagic-debug/blackmagic/blob/main/Makefile#L20).

Please let me know if there is a way to circumvent this issue.

lenvm, 2023-03-12:
I have been able to resolve the issue by adding the following lines to the `.bashrc` and `.zshrc` files, and subsequently sourcing these files.

```
export PATH="$HOME/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin:$PATH"
export PATH="/opt/homebrew/opt/make/libexec/gnubin:$PATH"
```

For reference how to modify the `.bashrc` file using the terminal, see [Install xPack GNU Arm Embedded GCC toolchain on Apple M1 Silicon](https://gist.github.com/mandrean/0194f6a05727161db74da1d6fffc075b#file-install-arm-embedded-gcc-toolchain-on-apple-m1-silicon-aarch64-arm64)

closed 2023-03-12 by lenvm

dragonmux, 2023-03-12:
We strongly suggest using the official ARM toolchain as the README and website instructions clearly say to, as we cannot make guarantees that the binaries from xPack are functional (this has been a problem) and we do not test with those compilers.

If you encounter further problems down the road, we suggest grabbing the toolchain [from ARM](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads), adding it to your $PATH in place of the xPack toolchain, running `make clean` and then building as before

lenvm, 2023-03-18:
@dragonmux, thank you for your suggestion to use the official ARM toolchain!

For future reference of others experiencing the same error:

I have installed the AArch32 bare-metal target (arm-none-eabi) from the [Arm downloads page](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) and after adding the following line to the `.zshrc` file I was able to build the code.

```
export PATH=$PATH:/Applications/ArmGNUToolchain/12.2.rel1/arm-none-eabi/bin
```