blackmagic-debug/blackmagic #1418: blackpillv2: PA1 is defined for both TDI and LED_UART

lenvm, 2023-03-20:
In the blackpillv2 platform code, PA1 is used for both TDI and LED_UART. Please find the references to the code:

- PA1 is defined as TDI in the [following lines](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/blackpillv2/platform.h#L52-L57).
- PA1 is defined as UART_LED in the [following lines](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/blackpillv2/platform.h#L76-L77).

This seems incorrect. Please correct me if I have overseen something.

dragonmux, 2023-03-20:
That seems incorrect to us too. The pins picked on that platform seem incredibly random and scatter-shot, but if someone with the board can weigh in about it, that would be incredibly helpful. 

Hopefully we can come to either a reason for the choice, or that it is a bug, soon and get something merged to either explain to the next person who reads that in 6mo or to fix it.

ALTracer, 2023-03-30:
I agree. LED_PORT_UART + LED_UART may lead to toggling PA1 as push-pull output when JTAG requires TDI (PA1) to be floating input. This may result in JTAG enumeration failures.
I swapped TDI for PA15 for my WeAct MiniF4 rev3.1 board just in case. While it does not visibly solve my problems with JTAG access, it is better than conflicting pin mappings.

lenvm, 2023-04-10:
As @dragonmux mentioned, the pinout looks a bit scatter-shot. The double usage of PA1 was also mentioned in [this issue](https://github.com/blackmagic-debug/blackmagic/issues/1048#issue-1268329983). To resolve the problem and to reduce the pin scatter I have come up with a proposal to reassign the pins.

Benefits:
- Pins are located closer to each other on the board.
- Pinout does not use PA13 and PA14, such that these pins stay available to program the board.
- No pin conflict between LED UART and TDI.

Before the details follow, let me post my questions:
- Does anyone foresee a problem with the proposed pinout, such as certain pins not working for the respective functions?
- If there are no problems foreseen, would it be ok to change the pinout as proposed?

Table 1: comparing current pinout to proposed pinout
| function | current (e490465) | proposal |
| -------- | ----------------- | -------- |
| nRst | PB4 | PB4 |
| TRST | PB5 (seems unused by code) | seems unused by code, thus not assigned |
| TDI | PA1 (also used by LED UART) | PB5 |
| TDO | PB3 | PB6 |
| TCK | PA14 | PB7 |
| TMS | PA13 | PB8 |
| PWR | PB8 | PB9 |
| UART TX | PB6 | PA2 |
| UART RX | PB7 | PA3 |
| USER KEY | PA0 | PA0 |
| LED UART | PA1 | PA1 |
| LED idle run | PC13 | PC13 |
| LED error | PC14 | PC14 |
| LED boot loader | PC15 | PC15 |

For information: please find below how the pins on the hardware are used by the current pinout, and by the proposed pinout.

[image]
Fig 1: Current pinout (commit e490465)

[image]
Fig 2: Proposed pinout

References:
- The images are modified versions of the [WeAct Studio Black Pill V2.0 pinout](https://docs.zephyrproject.org/3.2.0/_images/Blackpill_Pinout2.jpg).

ALTracer, 2023-04-10:
I like your proposal "in general". In specifics though, I'd change a couple things.
1. You still want to use PB5... and as digital float input (TDI), which is a non-5V-tolerant pin. If it was previously (TRST is output) unused, now it will be more susceptible to damage. Since this firmware bit-bangs anyway AFAIK, maybe better remap it elsewhere and make PB5 e.g. a LED.
2. Half of left side is actually occupied by USB FS PHY (PA11-PA12 mandatory, PA9-PA10 optional VBus sensing, OTG_ID albeit unconnected) and full SWJ-DP pins. It's good to be able to debug the BMP firmware itself, but some (very old) adapters like JLink v5 are incapable of SWD and can only perform old full JTAG. Is it possible to keep all of related pins in AF JTAG?
3. You seem to want to group the connectible pins together, or even better to one side/header. This is easily possible on e.g. RP2040 boards, and I think should be possible on WeAct F411 boards. Yes, there are Blackpill reset pin and PC13 LED, LSE pads and PA0 button, and even UART on PA2-PA3, but IMHO GPIOB bank/ADC is more promising. Is PWM input capture a requirement for SWO decoding btw? What about DMA? I only know that rp2040 adapters often use PIOSM for obscenely fast bitbanging SWD.

All in all, I get a feeling that the previous pinout was sometime copypasted from another platform, maybe even "native" or the venerable "swlink-bluepill". I see no benefits from keeping pinouts similar across platforms.

lenvm, 2023-04-13:
Hi @ALTracer, thank you for your response and inputs!

- Regarding 5V-intolerant pins: In my specific case I am actually using a [Black Pill V3.0](https://docs.zephyrproject.org/3.2.0/boards/arm/blackpill_f401ce/doc/index.html) with STM32F401CE, therefore all pins are 5V tolerant. Indeed, my objective in terms of pinout is to put the pins as close together as possible, near the bottom of the board. My specific case however does not apply to all users. I therefore think it makes sense for me create a pull request that allows the user to select alternative pinouts. The default pinout can take into account that the STM32F411Cx has 5V-intolerant pins. One of the alternative pinouts can match my objective to place the pins near the bottom of the board.
- Regarding using different pins than proposed: Since you seem to have a different objective in terms of pin placement, would you be able to provide me with a proposal for a pinout in line with the points in your comment? Then we can try to come to a conclusion on a desired default pinout, to first fix the issue of PA1 being defined for both TDI and LED_UART.

dragonmux, 2023-04-14:
The only thing we have to add at this time is that while nTRST is not presently used by the code, we still reserve the need to be able to use this in the future for future target support, so this pin still requires mapping.

There are plans to use the pin to help (when available on the target) ensure that the debug interface is in a known reasonable state when we go to do a bus scan, for example.

lenvm, 2023-04-15:
@dragonmux, thanks for your contribution! That nTRST will have a function in the future is very useful information to know!

I have come up with a new proposal for the default pinout. This takes into account the following points:
- nTRST must be defined.
- let's not use 5V-intolerant pins (PB5) for the default pinout.
- let's not use JTAG pins (PA13, PA14, PA15, PB3, PB4) for the default pinout.

Table 1: comparing current pinout to proposed pinout
| function | current (e490465) | proposal |
| -------- | ----------------- | -------- |
| nRst | PB4 | PA5 |
| nTRST | PB5 (optional) | PA6 (optional) |
| TDI | PA1 (also used by LED UART) | PB6 |
| TDO | PB3 | PB7 |
| TCK | PA14 | PB8 |
| TMS | PA13 | PB9 |
| PWR | PB8 | PA1 |
| UART TX | PB6 | PA2 |
| UART RX | PB7 | PA3 |
| LED UART | PA1 (also used by TDI) | PA4 |
| LED idle run | PC13 | PC13 |
| LED error | PC14 | PC14 |
| LED boot loader | PC15 | PC15 |
| USER KEY | PA0 | PA0 |
| USB VBUS | PA9 | PA9 |
| USB ID | PA10 | PA10 |
| USB D- | PA11 | PA11 |
| USB D+ | PA12 | PA12 |
| JTAG TMS | - | PA13 |
| JTAG TCK | - | PA14 |
| JTAG TDI | PA15 | PA15 |
| JTAG TDO | - | PB3 |
| JTAG TRST | - | PB4 |

[image]
Figure 1: proposed pinout

I look forward to your feedback (and possible go ahead from @dragonmux), such that I can create a pull request to implement this.

dragonmux, 2023-04-17:
That all looks good to us, so as long as other Blackpill v2 users give their thumbs up then we're happy to get a PR for this and merge it

closed 2023-04-22 by dragonmux