blackmagic-debug/blackmagic #1427: native: inconsistency in documentation and code definition of nRST pin

lenvm, 2023-03-30:
In the native platform code, the documentation of the code states that PA7 or **PC13** is used for nRST, depending on the hardware version. The code however defines that PA7 or **PA13** is used for nRST, depending on the hardware version.

Please find the references to the code:
- PC13 is defined as nRST for Hardware 6 and newer in the documentation, in the [following lines](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L71).
- PA13 is defined as nRST for Hardware 6 and newer in the code, in the [following lines](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L133-L134).

This seems incorrect. Either one or the other shall be altered. As I don't have access to the latest schematics, it would be great if someone could tell if PC13 or PA13 is correct.

dragonmux, 2023-03-31:
Hi lenvm, thank you very much for the bug report. 

nRST on hardware rev 6 is on pin PA9, with sense on PC13 as in the documentation. We will look at how to fix this properly and open a PR with the fix in the next few days. Most likely this explains why connect-under-reset has been a bit weird on the new hardware.

lenvm, 2023-03-31:
Thanks for the quick reply! Looking forward to seeing the fix.

As in the documentation section it states nRST is PC13, but actually nRST_sense is PC13, could you also please update `nRst` in [line 70](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L70) to something like `nRst_sense`, such that the documentation is fully consistent with the code?

lenvm, 2023-03-31:
After some additional code review, I've found two additional particularities in the native [platform.h](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h) file.

I'll add them here as the changes are similar to the original bug reported, but please let me know in case I shall open another issue.

AUX interface VBAT defined as:
- PA0 in the [documentation](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L102).
- PB0 in the [code](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L187).

TPWR defined as:
- PB0 in the [documentation](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L54).
- but not explicitly defined in the code. The only `GPIO0` defined in the code is `AUX_VBAT` in [line 187](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L187).

Therefore the comment regarding TPWR in the documentation seems redundant.

lenvm, 2023-04-03:
@dragonmux  Thanks for your elaborate reply!

The code changes in the branch `fix/native-pinout-mistakes` thus far (commit 2b5a16b) cover the concerns of this issue regarding nRST. Thanks for that! Updating `nRST_SNS` to `nRST_SENSE` would make it even clearer to me, as SNS might be interpreted as an undefined abbreviation.

- You are absolutely right about the VBAT port. I had overseen the definition of `AUX_VBAT_PORT` in [L176](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L176). In my opinion it would be clearer if the port definition was placed closer to the pin definition, so for instance directly above the definition of `AUX_VBAT`. Would that be an option?
- It would be great if you could update the code as described for `TPWR`, including a description of the relationship between `TPWR` and ADC1. From your explanation it is still unclear to me what is the relationship between `TPWR` and PB0, and where PB0 is defined in the code. I can only find that `PWR_BR` is defined as PB1 in [L129-L130](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L129-L130), as you mentioned. Can the block of code with comments that you mention describe the relationship with PB0 as well?
- Regarding Trace/SWO I will wait for the implementation to happen another time, as it is already on your list.

dragonmux, 2023-04-05:
We've written another round of clarifications. Do note that the pin mapping is explicit about what PB0 does even if the link to `PWR_BR` was not clear:
` * TPWR     = PB0  (input)  -- analogue on mini design ADC1, CH8`

Please let us know if things are clear enough now. We will open a PR to get these fixes and improvements merged into main in the mean time.

lenvm, 2023-04-05:
Hi @dragonmux, thanks for the additions to the branch and the respective pull request!

The changes in the pull request are great. It helps a lot to understand how the different pins are used.

I have also found out why it was unclear to me where PB0 was defined in the code. It isn't defined in the [platform.h](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h) file as I was expecting, but instead in [L270 of platform.c](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.c#L270). If I may add another suggestion, I suggest to add the pin definition of PB0 as a macro in platform.h, near the definition of PB1, and use that  in platform.c.

dragonmux, 2023-04-05:
Ah, yes, that's a fair call - and more consistent. We'll prod a suitable modification shortly then

lenvm, 2023-04-05:
Great, thanks!

One more question related to the pin definitions. I have seen that TRST is defined in [platform.h L127-L128](https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/native/platform.h#L127-L128). I however don't directly find where this macro is used in the code. Als I have uncommented the lines and build the code using `make PROBE_HOST=native` and found that the code compiles without problems. Could it be that the definition of TRST is not needed?

PR #1439 (2023-04-06): Fix: Native platform pinout mistakes

## Detailed description

There were a couple of pinout mistakes in the native platform and a lack of clarity about how certain pins related to their respective functions. This PR seeks to clarify these relationships and fixes the issues in the pinout.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this repository
* [x] It b [... cut]
closed 2023-04-06 by esden