blackmagic-debug/blackmagic #1440: Feature request: STM32H503/H562/H563/H573 support.

hydra, 2023-04-06:
STM Recently released the STM32H5 processors with a Cortex-M33 core.

https://www.st.com/en/microcontrollers-microprocessors/stm32h5-series.html

It would be awesome if BMP could support these processors.

I have NucleoH503 and H563 boards here for testing if needed.

I had a look at the flash chip registers, which are not the same as on previous MCUs.  It's dual-bank and has many security related features.

[image]

FWIW: I poked around in the OpenOCD code and tried a few things out there, and made some progress, and also posted on the OpenOCD mailing list but there has been zero replies to either my ML post or SF issue, here: https://sourceforge.net/p/openocd/tickets/389/  I'm hoping that perhaps BMP support might come sooner.

I'm primarily interested in the STM32H562/H563 MCUs (the H563 being mostly the same as a H562 + Ethernet)

hydra, 2023-04-06:
[image]

hydra, 2023-04-06:
Today I flashed BMD to a BaiteV2 F103CB board was able to get BMDA to scan the STM32H563.

Due to a different issue (#1442) I was unable to use the STLinkV3 on the NucleoH563 directly.

Here's the photos and log.

[image]

[image]

```
$ src/blackmagic.exe -tv 1
INFO: Open USB 18ec:5555 class ef failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
2 debuggers found!
Select with -P <pos> or -s <(partial)serial no.>
INFO: Open USB 18ec:5555 class ef failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
 1: DFCEC1E0, Black Magic Debug, Black Magic Probe (ST-Link/v2) v1.9.0-440-g6aa3a184
Black Magic Debug App v1.9.0-440-g6aa3a184
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 1d50:6018 DFCEC1E0 Black Magic Debug
 Black Magic Probe (ST-Link/v2) v1.9.0-440-g6aa3a184
Running in Test Mode
Target voltage: 3.60V Volt
Speed set to  3.2727 MHz for SWD
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
Trying old JTAG to SWD sequence
scan_multidrop: true
DP DPIDR 0x6ba02477 (v2 rev0) designer 0x43b partno 0xba
TARGETID 0x14840041 designer 0x20 partno 0x4840
RESET_SEQ succeeded.
AP   0: IDR=54770002 CFG=00000000 BASE=e00e0003 CSW=80000040 (AHB-AP var0 rev5)
ROM: Table BASE=0xe00e0000 SYSMEM=0x00000000, Manufacturer  20 Partno 484
 0xe00e4000: 0x00000000000a0000 Non ARM component ignored
ROM: Table END
AP   1: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=c3800040 (AHB-AP var0 rev8)
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, Manufacturer  20 Partno 484
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c9
 0 0xe000e000: Debug component - Cortex-M33 (System Control Space) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x2a04)
 -> cortexm_probe
CPUID 0x410fd214 (M33 var 0 rev 4)
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h7_probe
Calling stm32l0_probe
Calling stm32l4_probe
ID Code: 00004840
Calling stm32g0_probe
Please report unknown device with Designer 0x20 Part ID 0x4840
 1 0xe0001000: Debug component - Cortex-M33 (Data Watchpoint and Trace) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0xe0002000: Debug component - Cortex-M33 (Breakpoint Unit) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0xe0000000: Debug component - Cortex-M33 (Instrumentation Trace Macrocell) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 (Embedded Trace) (PIDR = 0x00000004001bbd21  DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0xe0042000: Debug component - Cortex-M33 (Cross Trigger) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x14 ARCHID = 0x1a14)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
 1 0xe0040000: 0x00000000 <- does not match preamble (0xb105000d)
2 Entry 0x1ff02002 -> Not present
3 Entry 0x1ff02002 -> Not present
ROM: Table END
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
***  1   Unknown ARM Cortex-M Designer 20 Part ID 4840 M33
```

hydra, 2023-04-06:
Datasheets:

* H562/H563 - https://www.st.com/resource/en/datasheet/stm32h562ag.pdf
* H573 - https://www.st.com/resource/en/datasheet/stm32h573ai.pdf
* H503 - https://www.st.com/resource/en/datasheet/stm32h503eb.pdf

Reference manuals:
* H562/H563/H573 - https://www.st.com/resource/en/reference_manual/rm0481-stm32h563h573-and-stm32h562-armbased-32bit-mcus-stmicroelectronics.pdf
* H503 - https://www.st.com/resource/en/reference_manual/rm0492-stm32h503-line-armbased-32bit-mcus-stmicroelectronics.pdf
Nucleo user manuals:
* H563 - https://www.st.com/resource/en/user_manual/um3115-stm32h5-nucleo144-board-mb1404-stmicroelectronics.pdf
* H503 - https://www.st.com/resource/en/user_manual/um3121-stm32h5-nucleo64-board-mb1814-stmicroelectronics.pdf

dragonmux, 2023-04-07:
We've implemented basic support for the STM32H5 series processor - see the linked issue above.

Please give the branch a test and let us know how you get on. We'll get implementing support for the option bytes in the next couple of days, and likewise support for the STM32H503 which uses a different ident code and possibly a slightly different memory map

dragonmux, 2023-04-09:
The linked PR does now theoretically support the STM32H503 as well. Sorting out target-specific commands for reading the UID, part revision, and handling the option bytes are our next thing to try to get done

hydra, 2023-04-11:
> We've implemented basic support for the STM32H5 series processor - see the linked PR above.
> 
> Please give the branch a test and let us know how you get on. We'll get implementing support for the option bytes in the next couple of days, and likewise support for the STM32H503 which uses a different ident code and possibly a slightly different memory map

I'll give things a try ASAP, hopefully later today.

nqbit, 2023-04-21:
When running `mon jtag_scan` from gdb the H5 can not be recognized. I get an unknown id 0x484. SWD does seem to work though. It seems like the masking also needs a shift for JTAG vs. SWD.

dragonmux, 2023-04-21:
nqbit: To be clear, this is while running the PR branch feature/stm32h5-stm32h7-support?

Could you please use a build that's been done with `ENABLE_DEBUG=1` (you'll need to run `mon debug en` before doing a JTAG scan if using firmware rather than BMDA, and you'll need to run with `-v 1` if using BMDA) and get a log of the scan phase execution? With the extra information that will provide we can then figure out what's going on with JTAG scan.

nqbit, 2023-04-21:
That is correct. I switched to the `feature/stm32h5-stm32h7-support` branch, commented out some LPC support, built for native, and DFU'd the application to the BMD.

I could get jtag scan to work by changing `#define ID_STM32H5xx 0x4840U` to `#define ID_STM32H5xx 0x484U`.

I can get logs to you in 4-5 hours.

dragonmux, 2023-04-21:
What that probably means is that the H5's JTAG interface is a DPv1 (its SWD interface is DPv2) so we're failing to read out DP_TARGETID which is causing the value dp->target_partno to not be set up and subsequently t->part_id is taking ap->partno instead of ap->dp->target_partno. That would also mean that the value is 0x484U is incidental (not reliable) but that also gives us the fix as it would make identification work in a similar way to how it does on STM32L4 type devices. Hopefully the logs corroborate that theory.

dragonmux, 2023-04-21:
We have updated the PR branch by switching target identification to using the DBGMCU_IDCODE value which should solve the identification problem for both JTAG and SWD.

nqbit, 2023-04-21:
Using the latest commits in the feature branch I get:

```
(gdb) mon swdp_scan
Target voltage: 2.7V
Please report unknown device with Designer 0x20 Part ID 0x4840
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x4840 M33
(gdb) mon jtag_scan
Target voltage: 2.7V
Please report unknown device with Designer 0x20 Part ID 0x484
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x484 M33
```

...building with debug now

nqbit, 2023-04-21:
`mon swdp_scan`

```
scan_multidrop: true
DP DPIDR 0x6ba02477 (v2 rev0) designer 0x43b partno 0xba
TARGETID 0x14840041 designer 0x20 partno 0x4840
RESET_SEQ succeeded.
AP   0: IDR=54770002 CFG=00000000 BASE=e00e0003 CSW=80000040 (AHB-AP var0 rev5)
ROM: Table BASE=0xe00e0000 SYSMEM=0x00000000, Manufacturer  20 Partno 484
 0xe00e4000: 0x00000000000a0000 Non ARM component ignored
ROM: Table END
AP   1: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=c3800040 (AHB-AP var0 rev8)
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, Manufacturer  20 Partno 484
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c9
 0 0xe000e000: Debug component - Cortex-M33 (System Control Space) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x2a04)
 -> cortexm_probe
CPUID 0x410fd214 (M33 var 0 rev 4)
 1 0xe0001000: Debug component - Cortex-M33 (Data Watchpoint and Trace) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0xe0002000: Debug component - Cortex-M33 (Breakpoint Unit) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0xe0000000: Debug component - Cortex-M33 (Instrumentation Trace Macrocell) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 (Embedded Trace) (PIDR = 0x00000004001bbd21  DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0xe0042000: Debug component - Cortex-M33 (Cross Trigger) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x14 ARCHID = 0x1a14)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M33 (Trace Port Interface Unit) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x11 ARCHID = 0x0000)
2 Entry 0x1ff02002 -> Not present
3 Entry 0x1ff02002 -> Not present
ROM: Table END
```

nqbit, 2023-04-21:
`mon jtag_scan`

```
Resetting TAP
Change state to Shift-DR
Scanning out ID codes
Return to Run-Test/Idle
Change state to Shift-IR
Scanning out IRs
Return to Run-Test/Idle
Change state to Shift-DR
Return to Run-Test/Idle
ID code 0x6ba00477: ADIv5 JTAG-DP port.
ID code 0x06484041: STM32 BSD.
DPv0 detected based on JTAG IDCode
RESET_SEQ succeeded.
AP   0: IDR=54770002 CFG=00000000 BASE=e00e0003 CSW=80000040 (AHB-AP var0 rev5)
ROM: Table BASE=0xe00e0000 SYSMEM=0x00000000, Manufacturer  20 Partno 484
 0xe00e4000: 0x00000000000a0000 Non ARM component ignored
ROM: Table END
AP   1: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=c3800040 (AHB-AP var0 rev8)
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, Manufacturer  20 Partno 484
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c9
 0 0xe000e000: Debug component - Cortex-M33 (System Control Space) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x2a04)
 -> cortexm_probe
CPUID 0x410fd214 (M33 var 0 rev 4)
 1 0xe0001000: Debug component - Cortex-M33 (Data Watchpoint and Trace) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0xe0002000: Debug component - Cortex-M33 (Breakpoint Unit) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0xe0000000: Debug component - Cortex-M33 (Instrumentation Trace Macrocell) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 (Embedded Trace) (PIDR = 0x00000004001bbd21  DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0xe0042000: Debug component - Cortex-M33 (Cross Trigger) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x14 ARCHID = 0x1a14)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M33 (Trace Port Interface Unit) (PIDR = 0x00000004000bbd21  DEVTYPE = 0x11 ARCHID = 0x0000)
2 Entry 0x1ff02002 -> Not present
3 Entry 0x1ff02002 -> Not present
ROM: Table END
```

dragonmux, 2023-04-21:
Hmm.. interesting - thank you; we'll dive into this properly tomorrow and try and get back to you with a better thought out solution that doesn't result in that happening.

dragonmux, 2023-05-29:
@nqbit sorry we didn't update this issue sooner, we did some work a few weeks ago which sought to address the issue you found, and which should work. Please do give the latest commit from the PR branch a try and let us know how you get on

PR #1444 (2023-06-22): Feature: STM32H5xx and STM32H7xx support

## Detailed description

This PR adds support for the STM32H5 series processors, and improves support for the STM32H7 series processors.

STM32H5 support includes support for Flash programming and mass erase, but does not yet include support for the option bytes

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this reposi [... cut]
closed 2023-06-22 by esden