blackmagic-debug/blackmagic #1442: Unable to open USB device "Failed to open ST-Link device 0483:374e - Access denied (insufficient permissions)"

hydra, 2023-04-06:
After both building locally (windows10/cygwin64) and being given up-to-date windows binaries by 'dragonmux' from discord I get the following error:

UCRT64:
```
$ ./blackmagic.exe -tv 1
INFO: Open USB 051d:0002 class  0 failed
Black Magic Debug App 2e88305
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:374e 002200184D4B500620373831 STMicroelectronics
 STLINK-V3
Failed to open ST-Link device 0483:374e - Access denied (insufficient permissions)
Are you sure the permissions on the device are set correctly?
```

Cygwin:
```
$ src/blackmagic -tv 1
INFO: Open USB 1022:149c class  0 failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
Black Magic Debug App v1.9.0-443-ge0b51965
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:374e 002200184D4B500620373831 STMicroelectronics
 STLINK-V3
Failed to open ST-Link device 0483:374e - Access denied (insufficient permissions)
Are you sure the permissions on the device are set correctly?
```

```
$ src/blackmagic.exe -l
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
No usable debugger found
No debugger found. Please check access rights to USB devices!
```

Turning on debug logs gives these files:

```
export LIBUSB_DEBUG=4
```

[command-tv_1.log](https://github.com/blackmagic-debug/blackmagic/files/11171655/command-tv_1.log)
[command-tv_1-cygwin.log](https://github.com/blackmagic-debug/blackmagic/files/11171656/command-tv_1-cygwin.log)

Here's the ProcessMonitor (https://learn.microsoft.com/en-us/sysinternals/downloads/procmon) log from the Cygwin version in CSV format:

[cygwin-process-log-full.zip](https://github.com/blackmagic-debug/blackmagic/files/11171659/cygwin-process-log-full.zip)

USB Drivers were upgraded, machine was rebooted several times.

[image]
[image]

The board being used to test is this:

https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-h563zi.html

Almost identical error occurs when using a NucleoH743ZI2 board (also with STLinkV3E)

```
$ src/blackmagic.exe -tv 1
INFO: Open USB 18ec:5555 class ef failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
Black Magic Debug App v1.9.0-440-g6aa3a184
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:374e 003600483137511639383538 STMicroelectronics
 ST-LINK/V3
Failed to open ST-Link device 0483:374e - Access denied (insufficient permissions)
Are you sure the permissions on the device are set correctly?
```

Both boards work fine otherwise, e.g. with STM32CubeProgrammer

[image]

Or with OpenOCD in the case of the NucleoH743ZI2 board.

hydra, 2023-04-06:
Further to this, I was able to Flash BMP (BMD these days?) to Baite V2 F103CB and installed WinUSB to the 2 USB interfaces using Zadig as below.

[image]

-->

[image]

[image]

[image]

And with BOTH the NucleoH563 (with STLinkV3E) and the Baite V2 (with BMP) I could run the `blackmagic.exe` and got this output:

```
$ src/blackmagic.exe -l
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
2 debuggers found!
Select with -P <pos> or -s <(partial)serial no.>
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
 1: DFCEC1E0, Black Magic Debug, Black Magic Probe (ST-Link/v2) v1.9.0-440-g6aa3a184
 ```

Subsequently connecting an H730 to the Baite V2 BMP gave me this:

```
$ src/blackmagic.exe -tv 1
INFO: Open USB 1366:0101 class  0 failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
2 debuggers found!
Select with -P <pos> or -s <(partial)serial no.>
INFO: Open USB 1366:0101 class  0 failed
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
WARN: libusb_get_string_descriptor_ascii() failed: Invalid parameter
 1: DFCEC1E0, Black Magic Debug, Black Magic Probe (ST-Link/v2) v1.9.0-440-g6aa3a184
Black Magic Debug App v1.9.0-440-g6aa3a184
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 1d50:6018 DFCEC1E0 Black Magic Debug
 Black Magic Probe (ST-Link/v2) v1.9.0-440-g6aa3a184
Running in Test Mode
Target voltage: 3.60V Volt
Speed set to  3.2727 MHz for SWD
scan_multidrop: true
DP DPIDR 0x6ba02477 (v2 rev0) designer 0x43b partno 0xba
TARGETID 0x04830041 designer 0x20 partno 0x4830
RESET_SEQ succeeded.
AP   0: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=c3800040 (AHB-AP var0 rev8)
Halt via DHCSR(01030003): success after 2ms
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, Manufacturer  20 Partno 483
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c7
 0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x00000004000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)
 -> cortexm_probe
CPUID 0x411fc272 (M7 var 1 rev 2)
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h7_probe
 1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004000bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M7 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb00e  DEVTYPE = 0x00 ARCHID = 0x0000)
 3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x00000004000bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
 4 Entry 0xfff41002 -> Not present
 5 Entry 0xfff42002 -> Not present
 ROM: Table END
1 0xe0041000: Debug component - Cortex-M7 ETM (Embedded Trace) (PIDR = 0x00000004001bb975  DEVTYPE = 0x13 ARCHID = 0x4a13)
2 0xe0043000: Debug component - CoreSight CTI (Cross Trigger) (PIDR = 0x00000004004bb906  DEVTYPE = 0x14 ARCHID = 0x0000)
3 Entry 0x1ff02002 -> Not present
ROM: Table END
AP   1: IDR=84770001 CFG=00000000 BASE=00000002 CSW=c3800040 (AHB-AP var0 rev8)
[... 51 lines cut ...]
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
SWD access resulted in no response
Recovering and re-trying access
SWD access resulted in no response
***  1   STM32H7 M7
RAM   Start: 0x00000000 length = 0x10000
RAM   Start: 0x20000000 length = 0x20000
RAM   Start: 0x24000000 length = 0x80000
RAM   Start: 0x30000000 length = 0x20000
RAM   Start: 0x30020000 length = 0x20000
RAM   Start: 0x30040000 length = 0x8000
RAM   Start: 0x38000000 length = 0x10000
Flash Start: 0x08000000 length = 0x100000 blocksize 0x20000
Flash Start: 0x08100000 length = 0x100000 blocksize 0x20000
```

So it's just the STLinkV3 on the Nucleo boards that I can't currently use.

sidprice, 2023-04-08:
@hydra @dragonmux I have tested the current "main" BMDA with a real STLInkV3 and I see the same problem you have experienced. I am currently trying to tack down the issue.

sidprice, 2023-04-15:
Just a quick note on this issue. While I am able to reproduce it I have not been able to track down the issue.

hydra, 2023-04-18:
@sidprice Thanks for the update, glad you're able to reproduce the issue.

jclab-joseph, 2023-04-21:
[image]
libusbK(3.1.0.0) is working.