blackmagic-debug/blackmagic #1448: Nordic nRF51,  Unsupported packet: qAttached & vCont

aguaviva, 2023-04-10:
After fixing #1394  a new issue arose, this is a bug for the new issue:

After flashing the firmware and issuing the command 'run' it hangs due to some unsupported packets, here are the logs

.gdbinit
```
target extended-remote :2000
monitor swdp_scan
attach 1
load microbit-samples-combined.hex
```
GDB output:
```
$ arm-none-eabi-gdb microbit-samples-combined.hex 
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
[ copyright stuff]
Reading symbols from microbit-samples-combined.hex...
(No debugging symbols found in microbit-samples-combined.hex)
Target voltage: 3.06V
Available Targets:
No. Att Driver
 1      Nordic nRF51 M0
0x0000d16c in ?? ()
Loading section .sec1, size 0x7c0 lma 0x0
Loading section .sec2, size 0xf000 lma 0x1000
Loading section .sec3, size 0x6918 lma 0x10000
Loading section .sec4, size 0x6438 lma 0x18000
Loading section .sec5, size 0x3874 lma 0x3c000
Loading section .sec6, size 0x20 lma 0x3fc00
Loading section .sec7, size 0x4 lma 0x10001014
Start address 0x0003c0c0, load size 130472
Transfer rate: 26 KB/sec, 952 bytes/write.
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: microbit-samples-combined.hex 
```
then it gets stuck at the last line

On the blackmagic side, in the log, this is what black magic reports when the hang happens

```
mdline: ><
ap_memread @ e000edf0 len 4: 03 00 03 00
ap_memread @ e000edf0 len 4: 03 00 03 00
ap_mem_write_sized @ e000ed0c len 4, align 4: 04 00 fa 05
ap_memread @ e000edf0 len 4: 03 00 03 03
ap_memread @ e000edf0 len 4: 03 00 03 00
ap_mem_write_sized @ e000ed30 len 4, align 4: 1f 00 00 00
Read CTRL/STAT: 0xf0000040
Write DP_ABORT: 0x00000000
DP Error 0x00000000
*** Unsupported packet: qAttached
*** Unsupported packet: vCont?
ap_mem_write_sized @ e000edf0 len 4, align 4: 01 00 5f a0
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 00 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 00
ap_memread @ e000edf0 len 4: 01 00 04 01
ap_memread @ e000edf0 len 4: 01 00 04 01
....
...
.
```
it seems it is polling for something that never happens, although that might be influenced by the commands `qAttached` and `vCont` fail

Note 2 things:

1- After uploading the firmware `comapre-sections' works fine
2- Rebasing with `fix/cortexm-halt-resume-correctness` I also get the same hang, here is the detailed debug info:

```
(gdb) set debug remote 1
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Sending packet: $vKill;1#6e...Ack
Packet received: OK
Packet vKill (kill) is supported
Starting program: /home/raul/repos/yotta/microbit-samples/build/bbc-microbit-classic-gcc/source/microbit-samples-combined.hex 
Sending packet: $vRun;#e6...Ack
Packet received: T05
Packet vRun (run) is supported
Sending packet: $qC#b4...Ack
Packet received: QC1
Sending packet: $qAttached#8f...Ack
Packet received: 
Packet qAttached (query-attached) is NOT supported
Sending packet: $Hg1#e0...Ack
Packet received: OK
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture> <feature name="org.gnu.gdb.arm.m-profile"><reg name="r0" bitsize="32"/><reg name="r1" bitsize="32"/><reg name="r2" bitsize="32"/><reg name="r3" bitsize="32"/><reg name="r4" bitsize="32"/><reg name="r5" bitsize="32"/><reg name="r6" bitsize="32"/><reg name="r7" bitsize="32"/><reg name="r8" bitsize="32"/><reg name="r9" bitsize="32"/><reg name="r10" bitsize="32"/><reg name="r11" bitsize="32"/><reg name="r12[508 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...Ack
Packet received: me="no"/></feature></target>
Sending packet: $qXfer:features:read:target.xml:416,3fb#b1...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC0070000FFFFFFFFD0060000000000C1C0070000FCFFFFFF00000000
Sending packet: $vCont?#49...Ack
Packet received: 
Packet vCont (verbose-resume) is NOT supported
Sending packet: $Hc0#db...Ack
Packet received: OK
Sending packet: $c#63...Ack
```

krzysztofgawrys, 2023-04-19:
@aguaviva this is STLinkV2 platform?

aguaviva, 2023-04-23:
yes it is.

closed 2023-11-13 by esden