blackmagic-debug/blackmagic #1466: Build error for hosted platform

lenvm, 2023-04-23:
I encountered a build error while trying to build commit cae88d6f on main for the hosted platform.

I used:
`make PROBE_HOST=hosted`

Returns:
```
 GEN      include/version.h
  CC      target/adiv5.c
target/adiv5.c:530:83: error: format specifies type 'unsigned short' but the argument has type 'unsigned int' [-Werror,-Wformat]
                                "Patching Designer code 0x%03" PRIx16 " -> 0x%03" PRIx16 "\n", designer_code, JEP106_MANUFACTURER_STM);
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
./target/jep106.h:60:42: note: expanded from macro 'JEP106_MANUFACTURER_STM'
#define JEP106_MANUFACTURER_STM          0x020U /* STMicroelectronics */
                                         ^~~~~~
include/general.h:107:41: note: expanded from macro 'DEBUG_WARN'
#define DEBUG_WARN(...)   debug_warning(__VA_ARGS__)
                                        ^~~~~~~~~~~
1 error generated.
make[1]: *** [Makefile:128: adiv5.o] Error 1
make: *** [Makefile:22: all] Error 2
```

dragonmux, 2023-04-23:
We wondered if that would be a problem when reviewing the format strings - you should be able to fix that by substituting `" PRIx16 "` for `u`, creating the string `"Patching Designer code 0x%03" PRIx16 " -> 0x%03u\n"`

if you want to submit a PR with that we'll be happy to merge it

dragonmux, 2023-04-24:
Ah, our bad - this is because you're building on macos and nobody adjusted serial_open() there - the patch you'll need to apply is to change all `return -1;` in the `#ifdef __APPLE__` branch to `return false`, and the return type to `bool`.

closed 2023-04-25 by dragonmux