blackmagic-debug/blackmagic #1469: pre-commit clang-format fails on Apple Silicon

lenvm, 2023-04-24:
I have tried to use the pre-commit on macOS on Apple Silicon. This failed.

After installing the pre-commit, and trying to commit, the following output is seen:

```
clang-format.............................................................Failed
- hook id: clang-format
- exit code: 250

dyld[5784]: Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: <xxx> /Users/xxx/.cache/pre-commit/repoxxx/py_env-python3.11/lib/python3.11/site-packages/clang_format/data/bin/clang-format
  Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)

check hex case...........................................................Passed
check accidental assignment..............................................Passed
```

The referenced paths do not exist:
- The path `/System/Volumes/Preboot/Cryptexes/OS/usr/` does not contain a folder `local`. It only contains a folder `lib`.
- The path `/usr/local/` does not contain a folder `opt`, nor `lib`.
- The path `/usr/lib/` does not contain the file `libzstd.1.dylib`.

I have installed zstd using:
```
brew install zstd
```

The path that contains the `libzstd.1.dylib` file is:
```
/opt/homebrew/Cellar/zstd/1.5.5/lib/
```

This is the expected path according to [Homebrew documentation](https://docs.brew.sh/Installation), as packages are installed in `/usr/local/` for macOS Intel and `/opt/homebrew/` for Apple Silicon.

After creating a symbolic link, to circumvent the difference between macOS Intel and Apple Silicon, using:
```
sudo mkdir /usr/local/lib
sudo ln -s /opt/homebrew/Cellar/zstd/1.5.5/lib/libzstd.1.5.5.dylib /usr/local/lib/libzstd.1.dylib
```

The following output is obtained:
```
clang-format.............................................................Failed
- hook id: clang-format
- exit code: 250

dyld[7487]: Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: <xxxx> /Users/xxx/.cache/pre-commit/repoxxx/py_env-python3.11/lib/python3.11/site-packages/clang_format/data/bin/clang-format
  Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)

check hex case...........................................................Passed
check accidental assignment..............................................Passed
```

Since it states `mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')` clearly something is not working for the Apple Silicon (arm64) architecture.

It is probably not directly an issue of the code on this repository, it is however an issue that someone may experience while trying to use this repository on Apple Silicon.

dragonmux, 2023-04-24:
This is a bug with Python 3.11 as packaged by whichever means you acquired it - this has absolutely nothing to do with BMD. From what we recall from the talk about BMDA about how M1/M2 silicon Homebrew lays things out, we'd expect the paths to refer into /opt/homebrew if it was a properly native binary.

lenvm, 2023-04-26:
@dragonmux This issue is solved now that #1471 is merged.

I will close it.

closed 2023-04-26 by lenvm