blackmagic-debug/blackmagic #1483: build fails on mac: command.c:34:10: fatal error: version.h: No such file or directory

xobs, 2023-04-29:
For some reason, I'm unable to build blackmagic on mac:

```
blackmagic on  imxrt-debug-format-fix 
❯ make V=1                                                                                                                                         (base) 
if [ ! -f libopencm3/Makefile ]; then \
		echo "Initialising git submodules..." ;\
		git submodule init ;\
		git submodule update ;\
	fi
/Library/Developer/CommandLineTools/usr/bin/make  -C libopencm3 lib/stm32/f1 lib/stm32/f4 lib/lm4f
rm -f .stamp_failure_lib_stm32_f1
  BUILD   lib/stm32/f1
/Library/Developer/CommandLineTools/usr/bin/make --directory=lib/stm32/f1 PREFIX="arm-none-eabi-" || \
		echo "Failure building: lib/stm32/f1: code: $?" > .stamp_failure_lib_stm32_f1
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f4
  BUILD   lib/stm32/f4
/Library/Developer/CommandLineTools/usr/bin/make --directory=lib/stm32/f4 PREFIX="arm-none-eabi-" || \
		echo "Failure building: lib/stm32/f4: code: $?" > .stamp_failure_lib_stm32_f4
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_lm4f
  BUILD   lib/lm4f
/Library/Developer/CommandLineTools/usr/bin/make --directory=lib/lm4f PREFIX="arm-none-eabi-" || \
		echo "Failure building: lib/lm4f: code: $?" > .stamp_failure_lib_lm4f
make[2]: Nothing to be done for `all'.
/Library/Developer/CommandLineTools/usr/bin/make  -C src
Git not found, assuming up to date include/version.h
  CC      command.c
arm-none-eabi-gcc -Wall -Wextra -Werror -Wreturn-type -Wno-char-subscripts -std=c11 -g3 -MD -I./target -I. -Iinclude -Iplatforms/native -Istm32/include -mcpu=cortex-m3 -mthumb -DSTM32F1 -DBLACKMAGIC -I../libopencm3/include -Iplatforms/stm32 -DDFU_SERIAL_LENGTH=9 -Os -DPC_HOSTED=0 -Iplatforms/common -c command.c -o command.o
command.c:34:10: fatal error: version.h: No such file or directory
   34 | #include "version.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [command.o] Error 1
make: *** [all] Error 2

blackmagic on  imxrt-debug-format-fix 
❯                                                                                                                                                  (base) 
```

xobs, 2023-04-29:
I've updated the title -- it appears as though the issue may be that I have blackmagic as a submodule.

In particular, if I manually set `GIT_VERSION=$(git describe --always --dirty --tags)` then it builds. However, by default it does not.

My Make-fu is not good enough to understand why. Even applying this patch does not result in a good build:

```patch
diff --git a/src/Makefile b/src/Makefile
index ed62836d..5f3becc9 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -145,7 +145,7 @@ endif
 .PHONY:        clean host_clean all_platforms clang-format FORCE
 
 
-GIT_VERSION := $(shell if [ -d "../.git" ]; then git describe --always --dirty --tags; fi)
+GIT_VERSION := $(git describe --always --dirty --tags)
 
 clean: host_clean
        $(Q)echo "  CLEAN"

```

dragonmux, 2023-04-29:
Ah, submodules don't have a .git dir, they have a .git file which has contents pointing to the real .git directory in the parent repo's. Please amend that if check to `-f` instead of `-d` and give it a try.

Regarding Make's behaviour, that's down to how := works, and is why the special CC variable logic at the top of the BMDA Makefile.inc:
https://github.com/blackmagic-debug/blackmagic/blob/62edb50e16c00f03a9294c679b3dd10cc3b3bf21/src/platforms/hosted/Makefile.inc#L1C1-L3

commit bd14624c (2023-04-29): make: Fix git detect when blackmagic is a submodule.

When the blackmagic repository is used as a submodule of another
repository the .git is a file redirecting to where the directory is
instead of being the directory itself. We need to test for "a file"
existing not for a directory.

Fixes #1483
commit 2fd391a9 (2023-04-29): make: Fix git detect when blackmagic is a submodule.

When the blackmagic repository is used as a submodule of another
repository the .git is a file redirecting to where the directory is
instead of being the directory itself. We need to test for "a file"
existing not for a directory.

Fixes #1483
PR #1484 (2023-04-29): Fix git detection when using blackmagic as a submodule.

## Detailed description

When the `blackmagic` repository is used as a submodule of another repository the .git is a file redirecting to where the directory is instead of being the directory itself. We need to test for "a file" existing not for a directory.

I have tested this by creating a mockup repository using this repository as a submodule. Now building within the submodule works again as intended.

Not sure if we need to backport this patch to the v1.9 branch and include in the next point release. It depends on what the liklyhood`  [... cut]
closed 2023-04-29 by dragonmux