blackmagic-debug/blackmagic #1485: IMXRT10xx Flash not initialised correctly when BOOTMODE[1:0]=2b00 and BT_FUSE_SEL=1

mubes, 2023-04-30:
When bootmode is 0 and BT_FUSE_SEL=1 the device will try to boot from the selected boot device (Pg 198 of imxRT1060 Proc Ref Manual Rev 3 07/2021). That doesn't end well at the moment;

```
$ blackmagic -tv5
<snip>
CPUID 0x411fc271 (M7 var 1 rev 1)
Calling kinetis_probe
dap_read_single failed (fault = 0)
dap_read_single failed (fault = 0)
Calling imxrt_probe
i.MXRT boot mode is 0
i.MXRT boot config is 00000000
-> booting from SPI Flash (FlexSPI)
Writing new instruction seqeunce to slot 0
0: 04 9f
1: 24 00
2: 00 00
3: 00 00
4: 00 00
5: 00 00
6: 00 00
7: 00 00
SPI Flash: mfr = 00, type = 00, capacity = 00000001
Writing new instruction seqeunce to slot 1
0: 04 5a
1: 08 18
2: 30 08
3: 24 00
4: 00 00
5: 00 00
6: 00 00
7: 00 00
8 byte SFDP read at 0x0:
    00 00 00 00 00 00 00 00
Flash size: 0MiB
  1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004000bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
  2 0xe0002000: Generic IP component - Cortex-M7 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb00e  DEVTYPE = 0x00 ARCHID = 0x0000)
  3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x00000004000bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
  4 Entry 0xfff41002 -> Not present
  5 Entry 0xfff42002 -> Not present
  ROM: Table END
<snip>
```
I _think_ when bootmode=0 the correct bootmode needs to be read from the BOOT_CFG1[7:4] fuses.

mubes, 2023-04-30:
Just to maintain a log...this is resolved when MCR0 is set to 0xffff8000, but other issues remain. To be continued.

dragonmux, 2025-03-06:
👋🏼 Is this still an issue present on `main` or is it resolved with the i.MXRT work that was done by other contributors? Given the closeness to the v2.0 release, we're going to bump this into v2.1 anyway, but it would be good to know before we do if this is actually resolved and closable instead.