blackmagic-debug/blackmagic #1488: RTT output hangs the BMP when not read by host

adamcbraun, 2023-05-03:
There seems to be a problem with RTT is the host is not reading the data out.

I have a target image that is running RTT and generates a fair amount of output. If I am running a monitor program on the RTT serial port, the output works well and there is no problem. Additionally if I do not enable RTT in the BMP, there is no problem. However, in the case where RTT is enabled but not being read on the host system, the BMP firmware hangs and becomes unresponsive.

This issues is caused because the first UART packet is loaded into the USB endpoint buffer and the buffer is never taken by the host. When the next rtt polling occurs, the rtt data is read but the rtt_write function gets stuck in the while loop attempting to write the packet to the endpoint buffer.  As the buffer has not been (and likely won't be) consumed by the host, this loop can never succeed and the device is hung. Further, because the BMP is stuck in this loop, it is unable to service the requests required to open the rtt serial port leaving the only recovery method to be a BMP power cycle.

koendv, 2023-05-26:
Does this patch solve the problem?
[patch.txt](https://github.com/blackmagic-debug/blackmagic/files/11575881/patch.txt)

A more complete solution would be to append the output of usb serial, traceswo decode, rtt, and redirect_stdout to a single circular buffer, and use the usb tx interrupt to trigger usb writes.

koendv, 2023-07-12:
You're right. You can use 10ms, and it will give the same result on BMF. Like this?
[patch.txt](https://github.com/blackmagic-debug/blackmagic/files/12028667/patch.txt)

> Is it possible to simultaneously redirect aux_serial, traceswo, rtt and semihosting stdout to the second CDC-ACM interface?

Yes. But now this results in four different usb packets to be sent, even if everything would fit in a single packet.

> Will it keep the order (of events happening, time-wise)?

There seems to be some code in aux_serial.c that does just that:

- implements a transmit fifo, therefore keeps order.
- uses USBUSART_DMA_TX_IRQ.

I think this would be an improvement: avoid polling. transmit less packets, but fuller packets if multiple data sources are used. Also only transmit a 0 byte packet if transmit fifo is empty and last packet was 64-byte.
But I'm not sure what the planning is?

koen

dragonmux, 2023-07-12:
Given things like RTT currently cause hangs if the aux serial port is not being monitored on the host, this would be a very interesting approach to us to see explored so if you'd like to explore it, and it does result in improvements in speed, or usability, etc, then we'd very much welcome a PR for it.

PR #1625 (2023-09-18): rtt: fix hang when aux serial not read

## Detailed description

This PR silently drops rtt output if the aux serial is unresponsive for 10ms.
Fixes #1488 RTT output hangs the BMP when not read by host.
Tested on blackpill-f411.

A more general solution would be to append the output of usb serial, traceswo decode, rtt, and redirect_stdout to a single circular buffer, and use the usb tx interrupt to trigger usb writes.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [ [... cut]