blackmagic-debug/blackmagic #1490: Add power down for NRF5x Debug Interface on disconnect

Miceuz, 2023-05-04:
I am working with NRF52 device and the problem I have is that BMP leaves Debug Interface Mode (DIF) powered on after disconnecting from target.

In [this resource ](https://devzone.nordicsemi.com/f/nordic-q-a/27255/resetting-dif-to-reduce-interface-current-on-nrf52) I found a set of J-Link commands that disable DIF:
```
SWDSelect
SWDWriteDP 1 0x04000000
exit
```
I wander if it's possible to implement this functionality in Black Magic Probe? I could create  PR if somebody would guide me to a correct track on how to implement this.

dragonmux, 2023-05-09:
👋🏼  The file that implements support for the nRF51 and nRF52 devices is https://github.com/blackmagic-debug/blackmagic/blob/main/src/target/nrf51.c

A change like this should probably get gated behind a target-specific monitor command because regaining debug access would then require a rescan which the user needs to know to do. However, such a change should be reasonably simple to achieve by defining a detach function that does the suitable memory write - for example:

```c
static void nrf51_detach(target_s *const target)
{
	/* Disable the debug interface to conserve power usage */
	target_mem_write32(target, 0x04000000U, 1U);
	cortexm_detach(target);
}
```

A PR for this would be most welcome w/ that fleshed out and tested against your target to ensure it does the right thing

Miceuz, 2023-05-10:
I have tried implementing the way below, but unfortunately it does not work, the target still is stuck in high power consumption with DIF enabled after detachment. But I have noticed you detach first and disable the interface later in your example - this seems counter intuitive, but I am not  familiar with DP at all, so maybe this makes sense. Will try this sometime.

I have created a custom detach function:

```c
static void nrf51_detach(target_s *t) {
  adiv5_access_port_s *ap = t->priv;
  adiv5_dp_write(ap->dp, MDM_POWER_EN, 0x04000000);
  adiv5_dp_write(ap->dp, MDM_POWER_EN, 0x04000000);
  cortexm_detach(t);
}
```

and mapped it here:
```c
bool nrf51_mdm_probe(adiv5_access_port_s *ap)
{
	switch (ap->idr) {
	case NRF52_MDM_IDR:
		break;
	default:
		return false;
	}

	target_s *t = target_new();
	if (!t)
		return false;

	t->mass_erase = nrf51_mdm_mass_erase;
	adiv5_ap_ref(ap);
	t->priv = ap;
	t->priv_free = (void *)adiv5_ap_unref;

	t->detach = nrf51_detach;

	uint32_t status = adiv5_ap_read(ap, MDM_PROT_EN);
	status = adiv5_ap_read(ap, MDM_PROT_EN);
	if (status)
		t->driver = "Nordic nRF52 Access Port";
	else
		t->driver = "Nordic nRF52 Access Port (protected)";
	t->regs_size = 0;

	return true;
}
```

Miceuz, 2023-05-11:
Now when working on this I have stumbled on another thing. I see two entries for nRF52 when doing swdp scan:
```
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
```

If I attach to 1, I appearently get Cortex M4 target and my detach function is not called upon detaching. But if I attach to 2 something goes wrong between gdb and bmp:
```
(gdb) attach 2
Attaching to Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
Truncated register 0 in remote 'g' packet
```
I no longer see nRF52 specific commands in `monitor help`:
```
(gdb) mon help
General commands:
	version -- Display firmware version info
	help -- Display help for monitor commands
	jtag_scan -- Scan JTAG chain for devices
	swdp_scan -- Scan SW-DP for devices: [TARGET_ID]
	auto_scan -- Automatically scan all chain types for devices
	frequency -- set minimum high and low times: [FREQ]
	targets -- Display list of available targets
	morse -- Display morse error message
	halt_timeout -- Timeout to wait until Cortex-M is halted: [TIMEOUT, default 2000ms]
	connect_rst -- Configure connect under reset: [enable|disable]
	reset -- Pulse the nRST line - disconnects target: [PULSE_LEN, default 0ms]
	tdi_low_reset -- Pulse nRST with TDI set low to attempt to wake certain targets up (eg LPC82x)
	tpwr -- Supplies power to the target: [enable|disable]
	traceswo -- Start trace capture, Manchester mode: [decode [CHANNEL_NR ...]]
	heapinfo -- Set semihosting heapinfo: HEAPINFO HEAP_BASE HEAP_LIMIT STACK_BASE STACK_LIMIT
Target specific commands:
	erase_mass -- Erase whole device Flash
	erase_range -- Erase a range of memory on a device
```

What is going n here?

dragonmux, 2023-05-11:
The MDM ("nRF52 Access Port") is the recovery interface for the processor specifically and not the general purpose debug interface. This is what `nrf51_mdm_probe()` is creating and handling

Miceuz, 2023-05-11:
right, got it, thanks, I'm trying your proposed solution right now, looks promising.