blackmagic-debug/blackmagic #1492: Potential regression in hosted mode bluepill->bluepill

mean00, 2023-05-08:
Hi
While updating my local copy , i noticed a potential issue 
How to reproduce :

1- Connect BMP running on a bluepill to another blupill
Load a random blink.elf sw directly (gdb->/dev/ttyACMx)
It works

2- Now go through hosted mode, using the same sw and the same BMP host built from the same source,  running on the same bluepill
( updated 10mn ago to be sure it was not transient, _v1.9.0-693-g6c193f06) )

load blink.elf
The memcompare fails for every block written

Section .text, range 0x8000000 -- 0x8004750: MIS-MATCHED!
Section .ARM.exidx, range 0x8004750 -- 0x8004758: MIS-MATCHED!
Section .data, range 0x8004758 -- 0x80047a0: MIS-MATCHED!
Section .rodata, range 0x80047a0 -- 0x8004d1c: MIS-MATCHED!
Section LNVECTOR, range 0x8004e00 -- 0x8004f30: MIS-MATCHED!

Indeed, The flash has been erased but not written

Additionaly, the hosted bmp sw is complaining 

scan_multidrop: false
remote_v3_adiv5_mem_write_bytes error around 0x08000000
remote_v3_adiv5_mem_write_bytes error around 0x08000400
remote_v3_adiv5_mem_write_bytes error around 0x08000800
remote_v3_adiv5_mem_write_bytes error around 0x08000c00
remote_v3_adiv5_mem_write_bytes error around 0x08001000
remote_v3_adiv5_mem_write_bytes error around 0x08001400
remote_v3_adiv5_mem_write_bytes error around 0x08001800

NB:  It works with RP2040 target
It seems to work if the same BMP sw is running on a bluepill and the target is a rp2040, both in direct and in hosted mode.

It used to work a couple of months back with the same setup

Sorry for the noise if it is pbcak

dragonmux, 2023-05-08:
Could you please get a log of BMDA running `src/blackmagic -v 13 > stm32f1-bmda-bmp.log` and post that on this issue report as this looks like a bug in the v3 remote protocol implementation, possibly introduced with #1459 which overhauled that system.

That should allow us to figure out what exactly's going wrong to analyse this and come up with a fix.

mean00, 2023-05-09:
Please find the log file attached
Thank you
[stm32f103-bdma-bmp.log](https://github.com/blackmagic-debug/blackmagic/files/11427378/stm32f103-bdma-bmp.log)

mean00, 2023-05-09:
I did a bit of bissection (it is not accurate but may help) : 
a3aa29cdc17598e90b59a4121cee2231199b9acf   => works
6730405b437d198beddefa2d2932217126b88963 => does not work

dragonmux, 2023-05-09:
> Please find the log file attached

Thank you very much!

> Thank you
> [stm32f103-bdma-bmp.log](https://github.com/blackmagic-debug/blackmagic/files/11427378/stm32f103-bdma-bmp.log)

Unfortunately, while this helps a lot in figuring out what was going on, we goofed on what verbosity level to request - sorry to ask if you could re-capture the log changing `-v 13` to `-v 29` which will enable additional diagnostics and complete tracing of the remote protocol interactions

mean00, 2023-05-09:
Please find attached the log using -v 29
Thank you
[stm32f1-bmda-bmp-v29.log](https://github.com/blackmagic-debug/blackmagic/files/11434125/stm32f1-bmda-bmp-v29.log)

dragonmux, 2023-05-31:
Sorry it's been so long getting back to you about this, @mean00! We started trying to dive the logs you so kindly provided, but they appear to be truncated slightly (not your fault) as the error that caused the `remote_v3_adiv5_mem_write_bytes error around 0x08000000` lines was not getting reported and logged. We've addressed this on the branch fix/bmda-improved-remote-protocol-error-handling, so please give that branch a try using `-v 61` as the logging level and let us know how you get on please.

From the data available, it looks like a firmware remote protocol bug but till we get a bit more data from the remote protocol transactions we cannot tell the nature of the bug.

mean00, 2023-06-10:
Confirmed to be fixed, thanks & sorry for the slow reply

dragonmux, 2023-06-10:
No worries, thank you for getting back to us to confirm that!

PR #1501 (2023-06-02): Fix: Remote protocol write length alignment error

## Detailed description

This PR addresses a regression caused by #1459 where BMDA can calculate a memory write transfer length inappropriate to the alignment required for that write. In the process, this also improves the error reporting of the BMDA remote protocol implementation to make debugging in the future easier.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/b [... cut]