blackmagic-debug/blackmagic #1497: make undecoded traceswo port accessible without raw USB access

mbenkmann, 2023-05-23:
There seem to be difficulties in accessing the raw data from traceswo:

https://github.com/Marus/cortex-debug/issues/20

because it requires raw USB device access. I tried to find out why BMP, when using undecoded traceswo does not expose a COM port that can be accessed and the only reason I found is because the data is binary. If that is indeed the only reason, I would like this to be reconsidered. I'm not aware of any requirement that data on a COM port be human readable, at least not on Linux. Is this a Windows problem?

Maybe a parameter could be added to "monitor traceswo" that exposes the raw port as a COM port, so that at least on operating systems that do not suffer from such restrictions, access does not require libusb.

An alternative could be to add a parameter to "monitor traceswo" that sends the raw SWO data in base64-encoding to the COM port. That way the data on the COM port is safe but still contains all of the SWO information. And base64 is easily reversed on the reading side.

I'm sure with either solution it would be a lot easier for SWO data consumers like the above-referenced cortex-debug to access the SWO data from BMP.

ALTracer, 2024-06-02:
> requires raw USB device access

Only Javascript (nodejs?) has those difficulties, but they can access native bindings for e.g. [serialport](https://github.com/serialport/node-serialport), so I fail to see the problem with using libusb1. Every existing consumer of Trace Capture Interface uses libusb1, see: in-tree [swolisten.c](https://github.com/blackmagic-debug/blackmagic/blob/main/scripts/swolisten.c#L441), another project [Orbuculum](https://github.com/orbcode/orbuculum/blob/main/Src/orbtraceIf.c#L467), another project [Black Magic Trace](https://github.com/compuphase/Black-Magic-Probe-Book/blob/main/source/swotrace.c#L1476); technically both OpenOCD and texane/stlink st-trace also use raw bulk access against STLINK/V2, but it's less related. See also JLink and CMSIS-DAP.
All aforementioned programs are written in C and compiled to native code. You would be better looking at pyocd, which is written in an interpreted language (Python), but is also a gdbserver and deals with libusb1 and raw commands for every USB adapter https://github.com/pyocd/pyOCD/blob/main/pyocd/trace/swv.py#L119

> any requirement that data on a COM port be human readable

No such thing (assuming software flow control via XON/XOFF characters is disabled). The very same Orbuculum will happily capture Async/UART encoded TraceSWO from a plain USB-UART bridge, or even CDC-ACM soft bridges, including STLink & JLink & BMP VCP interfaces.

Users of Manchester encoded streams are in a pickle here, because this is only captured by BMP `native` platform on Trace Capture Interface (and a much more expensive Orbtrace Mini), there's no cheese UART way around it. (Usecases: dynamic clocking & sleep states, unknown/incompatible baudrate.)

> sends the raw SWO data in base64-encoding to the COM port

Data can be sent raw, or in case of Async encoding, you can just use the Aux UART Rx pins directly and have your raw data. Hexify/unhexify as part of GDB RSP (and BMPremote) already introduces overhead, and there's no base64 codec in firmware, which will take up code space.

>  exposes the raw port as a COM port

USB descriptors of Blackmagic Debug Probe are not dynamic, there's only one configuration. Moreover, every CDC-ACM requires up three endpoints per ACM specs and desktop OS kernel drivers' implementations. What you're suggesting is a decent complexity refactor of traceswo (and usb-) related parts. I also have to state that STM32F4-based BMP-compatibles cannot provide more than EP 3, they already struggle with two CDC-ACM and a Trace Capture interface, a third ACM is not a good option -- related Notification / Interrupt endpoints would lie on EP4, EP5, EP6 -- this is 7 IN endpoints, counting EP0.

If users only need ITM printf, then `monitor traceswo decode` will stop dumping raw traffic on Interface 5 and start dumping decoded ITM traffic of all or selected channels onto Aux CDC-ACM serial port. Hardware events (from DWT) are dropped.

I see the solution this way: Cortex-Debug may want to spin up not just gdb, but also some proxy sink server, like ~~st-trace~~ or ~~OpenOCD tpiu~~ or `./scripts/swolisten -d` or `./orbuculum` and attach to its TCP socket for acquiring raw SWO data. Of course, they don't do configuration and start/stop of capture, BMP implements `monitor traceswo` command for this on GDB RSP port. Yes, it's an extra native code dependency, no, I'm not qualified enough to think of a better solution -- maybe rewrite swolisten.c in JS with libusb1 bindings?

CC @mubes @dragonmux

dragonmux, 2024-09-13:
> I tried to find out why BMP, when using undecoded traceswo does not expose a COM port that can be accessed and the only reason I found is because the data is binary. If that is indeed the only [...]

It being binary is not the reason at all. The reason is threefold:

1. USB CDC ACM requires at minimum 2 endpoints (3 if implementing it properly spec compliant) - 1 bulk in, 1 bulk out (and for full compliance, 1 interrupt in) - over two interfaces; we would be ignoring the bulk out interface and we are already extremely short on endpoints w/ us already using all the available endpoints on platforms that use older DWC2's such as platforms based on the STM32F4 series chips. This just wouldn't work out at all, particularly as ignoring one of the in or out endpoints is liable to lock up software expecting an actual serial interface.

2. It would entirely break existing software built on the existing interface. USB CDC ACM interaction takes a different path through the OS and tangos with all kinds of driver shenanigans, one of which includes the entire inability to determine on Windows how many bytes are waiting on the interface to dequeue, which introduces needless slowness and complexity to handle (BMDA already suffers with this, we would not wish this on anyone).

3. Not using a serial interface, noting that it is mono-directional (device -> host only), greatly simplifies things in the firmware, particularly as the way that transmission control on USB CDC ACM works via EP0 would make for monumental headaches in the SWO implementations when they reach buffer limits. This allows us to implement handling for overflow conditions more correctly. It has a secondary benefit of not borking the user's console if they choose to connect software to the interface that is expecting displayable data, but this is not the primary concern.

With regards to "Is this a Windows problem?" actually, it's a macOS problem in regards to one of the OS shenanigans point 2 makes - can't unbind the native serial driver there, have to go through that to access the interface, and all the transmission control headaches that causes. Windows is a problem in regards to determining how much data can be read off the interface efficiently, but at least `ReadFile()` works there even if not a pleasant situation.

Anything that increase the bandwidth requirements is also out because SWO is already bandwidth staved getting the data to the host sensibly. That means no Base64 (x1.(3) on data requirements, not that we have spare CPU cycles for this really), nor any hex encoding (2x on data requirements). We are already dropping some data when recovering in Manchester mode at 115200 baud due to the bandwidth problem.

Finally, we would point out that node.js has several USB bindings libraries, [including libusb bindings](https://github.com/node-usb/node-usb), and potentially WebUSB which is included in the same node-usb library linked. This gives ample options for Cortex-Debug to integrate direct USB communications with the SWO data interface. We provide WinUSB bindings descriptors in the MS OS Descriptors for the firmware so a suitable driver for this is automatically bound by the kernel.

closed 2025-07-25 by dragonmux