blackmagic-debug/blackmagic #1498: stlinkv2: rom overflow, more details needed on how to create a new platform branch

aguaviva, 2023-05-25:
I read `Flashsize_F103` but I am not sure on how exactly proceed with this option:

```
 Branch STlink V2-1 as separate platform and and care for the original 
 STlink platform by restricting/omitting features.
```
I'd be great to have more details, any chance you could elaborate a bit more?

BTW This is the exact error I am getting (just in case it helps others)

```
[...]
  CC      platforms/common/usb_dfu_stub.c
  CC      platforms/common/aux_serial.c
  LD      blackmagic.elf
/opt/arm-gnu-toolchain-12.2.mpacbti-rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: address 0x80215a0 of blackmagic.elf section `.text' is not within region `rom'
/opt/arm-gnu-toolchain-12.2.mpacbti-rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: blackmagic.elf section `.ARM.exidx' will not fit in region `rom'
/opt/arm-gnu-toolchain-12.2.mpacbti-rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: address 0x80215a0 of blackmagic.elf section `.text' is not within region `rom'
/opt/arm-gnu-toolchain-12.2.mpacbti-rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 6072 bytes
Memory region         Used Size  Region Size  %age Used
             rom:      137144 B       128 KB    104.63%
             ram:       12020 B        20 KB     58.69%
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:125: blackmagic.elf] Error 1
make: *** [Makefile:17: all] Error 2
```

Thanks for this awesome tool

dragonmux, 2023-05-25:
The error you're seeing is an artefact of just how close to completely out of Flash the project has become. You can disable a couple of targets you don't presently care about by deleting (not commenting out!) their entries from src/Makefile in the big `SRC +=` list.

Just an FYI, the compiler you're using is a beta release compiler, so if something has miss code generated or gone wrong beyond that error and you result in non-booting firmware, we recommend you grabbing the 12.2.rel1 release toolchain from ARM which is what we support using. We're not saying that is the problem in this case, but just is something of note.

aguaviva, 2023-05-26:
Thanks for both tips, I managed to get all in the flash and with the right compiler(I had not noticed it was a beta!) :)

closed 2023-05-26 by aguaviva