blackmagic-debug/blackmagic #1500: Clarification on documentation for st-link baite

marianomd, 2023-05-31:
I have flashed bmp on my baite st-link.

Compile command used: 

```
make PROBE_HOST=stlink SWIM_AS_UART=1
```

There are two issues that I couldn't find in the documentation:

1) I can receive serial through RST, but cannot send throguh SWIM. Is this because it is pulled high? Should I desolder the pullup resistor?

2) None of the two leds are lit. The blue one is connected to PA9. Is this a problem with revision auto-detection, or is it some other problem?

Thanks

dragonmux, 2023-05-31:
There was an issue we saw on the track ~ 1½ years ago which had pictures of a style of "Baite" clone dongle, with the term coming from text stamped into the plastic case, but nobody's ever actually defined what makes a dongle a "Baite" dongle.

This looks like a reasonably standard ST-Link v2 clone, meaning building with `make PROBE_HOST=stlink ST_BOOTLOADER=1` should have given functional firmware, but it's a huge hodgepodge with a lot of variability, so it might be worth checking that the header pins go to the expected processor pins, especially given what you said about the LEDs.

`SWIM_AS_UART=1` makes the SWIM pin part of a UART interface, but we don't recall from the contribution that added that if it requires removing a resistor or not - as you can't transmit from the probe, this would suggest that removing that resistor is likely required, so if that's not too difficult for you, it's worth a shot.

Sorry that we can't be more directly helpful.

marianomd, 2023-05-31:
Thanks a lot. I will check the pins.

Regarding the bootloader, I wiped the ST one and flashed blackmagic_dfu , that works like a charm. (Had to buy a second st-link. Before that I tried to ovewrite the bootloader with my own code but it seems the protection kicked in)

closed 2023-05-31 by marianomd