blackmagic-debug/blackmagic #1520: LPC546xx flash loading broken

dlaw, 2023-06-21:
```
$ gdb-multiarch 
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC54608J512 M4
 2      LPC55 Debug Mailbox 
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000006e8 in ?? ()
(gdb) load bin/bootloader.elf
Error erasing flash with vFlashErase packet
```

Broken in 026193ad82ca88433c668529847e0392bd3b5fc1, working in c5b7075477845278f0d5c7182f818616ebb72ae5 (with a gdb rollback to <= 10). I will bisect when I have a little more time available.

dragonmux, 2023-06-21:
Thank you for the issue report - we look forward to hearing more when you get a chance, or possibly spending some time ourself on it if we can locate a suitable target to tests against.

It's quite notable in the output you've shown that despite not being an LPC55 part, you're seeing the LPC55 support trigger and provide a second target, so that could be indicative of part of the problem.

dlaw, 2023-08-28:
With some initial debugging I've found three different behaviors:

c4869a54733ae92099a7316954e34d1ab7b6097c: works

993c74bef471657981467fe312dfcee45d9da44d: fails with infinite hang on load command (this was a bug which also affected LPC8xx targets)

c7c209ac7c798015cbbe2fead198f838dfd8a769: fails as described above (Error erasing flash with vFlashErase packet)

Bisecting on the exact point of failure is tricky due to the other bug which covers a bunch of the middle commits, as well as a very large LPC refactor made of many commits (which I would imagine is what caused the regression).

If there is a developer with time to look into this, please let me know -- happy to provide hardware.

dragonmux, 2023-08-29:
`mon erase` meaning `mon erase_mass` we assume - this runs entirely different routines to `mon erase_range`, which uses the `target_flash_erase()` pathway to erase Flash blocks - this suggests that it's specifically a regression in `lpc546xx_flash_erase()` or one of the supporting routines.

We'll check if we did actually get a LPC546xx board in since the original report, and see about dissecting this code now we have some commit ranges to look at - see if we can spot something obvious that would lead to a behaviour change, etc.

dlaw, 2023-08-29:
Yes. `mon erase_mass`. I corrected my comment above.

I'd be happy to test any other probe firmware versions, if you have any ideas to try.

dragonmux, 2023-12-31:
👋🏼 Currently checking through the bug tracker and we're wondering if this is still broken or if after the work that was done to the LPC families support in v1.10 this is now actually fixed?

dlaw, 2023-12-31:
Still broken in v1.10. Please let me know if there's anything you would like tested.

dragonmux, 2023-12-31:
If you could do a build of BMDA (both v1.10.1 and latest main) and get us some logs using `src/blackmagic -v 13` (info, target and protocol levels enabled) that would be fantastically helpful to know where things are going wrong and start triaging this.

dlaw, 2024-01-19:
Sorry for the delay. I haven't forgotten about this and I will collect these logs soon!

dlaw, 2024-01-22:
I tested with blackmagic hosted attached to a jlink probe. Logs attached. Here is the GDB session corresponding to these logs:
```
$ gdb-multiarch
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote localhost:2000
Remote debugging using localhost:2000
(gdb) mon swdp_scan
Target voltage:  3.303V
Available Targets:
No. Att Driver
 1      LPC546xxJ512 M4
 2      LPC55 Debug Mailbox 
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x00000682 in ?? ()
(gdb) load bin/A027
Display all 179 possibilities? (y or n)
(gdb) load bin/A0275PAM00.bootloader.elf
Error erasing flash with vFlashErase packet
(gdb) 
```
[blackmagic_logs.txt](https://github.com/blackmagic-debug/blackmagic/files/14016429/blackmagic_logs.txt)

DxRT, 2025-09-04:
Hi,
I have this issue with current release flashed on Black Pill (I tested Probe on stm32 and it works). I really, really need this to work because I was left without any probe (you know, loose cable with 24V) 😂

dragonmux, 2025-09-05:
Looking through lpc546xx.c, we don't actually see that part ID in the LUT it uses:
https://github.com/blackmagic-debug/blackmagic/blob/1e3bcd8accabaef0f137f6d39d011cb6177b7ce4/src/target/lpc546xx.c#L91-L103

So, our first thought here because that means the Flash and RAM Tables aren't being set up correctly.. would be to add your part to that LUT w/ a string for which size of Flash it has and definitions for Flash and RAM. From there, we can then untangle the Flash IAP call code for this part and try and figure out why the LPC generic routines are faceplanting.

ALTracer, 2025-09-05:
> LPC546xx: Part [2] ID 0x4294264328

I'm pretty sure this is a cosmetic bug, u32 4294264328 is 0xfff54608. Should've been PRIx32. I guess I did it in PR1522, I can fix that. But I should not have affected actual flash driver logic.
https://github.com/blackmagic-debug/blackmagic/blob/1e3bcd8accabaef0f137f6d39d011cb6177b7ce4/src/target/lpc546xx.c#L140

I've read the source diagonally and 0x03000205 is IAP entrypoint, 0x03000204, plus Thumb bit, so your chip hardfaults as soon as BMP tells it to run something IAP (likely a flash op command). No further ideas without access to docs. Additionally, I would recommend opening a separate issue, but it's optional.

DxRT, 2025-09-06:
Yeah, it's cosmetic, I figured it out.
Hard fault is caused by probe connected to MCU, and I'm sure, because I have load on it and when disconnect probe, program runs without. That adress is end of 1st level bootloader on that MCU and usually place where "rau and halt" will stop, so that one step should be added. Issue is that is something with sending commands to mcu and all are invalid. LPC have that weird sync procedure for debuging and maybe here is issue that first one fails? and then all coms are broken, idk.
I wanted to do quick sweep with hope that issue is simple, so whenI saw this open issue I wanted to contribute because I have a bit time and that boards.
Anyway, in Monday I'll get dedicated probe, so when I find a bit time, I'll test with analyzer what is going on on SWD BMP and dedicated, so we might fix that for BMP (I'll open new issue then). 
Cheers! And keep doing good work here ;)