blackmagic-debug/blackmagic #1535: sttlink v2, SWIM_AS_UART, how to use it?

aguaviva, 2023-07-11:
1) What baudrate do I have to use? 
2) What is the RX pin in the swim connector, the "swim"or the "reset" pin?

Cheers

dragonmux, 2023-07-11:
Hi aguaviva, regarding the baud rate - this is set when you open the serial port on the host side and can be any value up to the max supported by UART1 on the STM32F1.

The pinout not being noted in the platform README is a bit remiss, but the remapping puts the UART on pins PB6 (TX) and PB7 (RX) as seen by the processor. Googling "stlink SWIM schematic" gives us results showing that the SWIM line should be going to PB7 and reset to PB6, making SWIM the receive pin.

aguaviva, 2023-07-11:
Thank you, you are awesome.