blackmagic-debug/blackmagic #1554: nRF52 debug port left active after debug session

jbussmann, 2023-07-20:
I am working on a nRF52840 based low power application and I observed excessive power consumption. I narrowed down the issue to the debug port being left active. The issue always occurs after using a Black Magic Probe to download firmware and vanishes after a power cycle. According to section 3.2 of [this post](https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/serial-wire-debug-port-interface-for-nrf52832) at nRF DevZone the debug port can also be disabled by clearing the Control/Status Register (CTRL/STAT).

The power cycle is problematic since the application requires a permanently connected power supply and doesn't provide simple means for cycling power. 

- Is it expected behaviour to leave the debug port active after a debug session? If not, how do I achieve expected behaviour?
- Is there a way to send specific SWD commands, like the one mentioned above?
- Would it be possible and reasonable to send this command, whenever Black Magic Probe detaches from the target?

dragonmux, 2023-07-29:
Hi jbussmann,

1) Yes, it is expected behaviour that BMD will leave the DP in an active state post-session. While the DP can be told to power the APs and debug bus back down, this is actually a problematic headache for the project as many users expect after disconnecting from a core, to be able to reconnect or connect to another on the scan chain, and switching off the debug domain prevents any of that working. GDB doesn't provide an inbuilt mechanism to say "I'm done now!".
2) There is not a way to request specific commands be sent. For what it's worth, it is /not/ a SWD command, but an ADIv5 DP write (ADIv5 is protocol agnostic at this layer)
3) As explained in 1), it is not possible from just detaching. However, it is possible to add a monitor command that requests BMD to throw away the target list and tears all debug interfaces back down - something like `monitor teardown` or something similar. If this would work for your needs, we can look at implementing it as a patch.

zyp, 2023-07-29:
For what it's worth, I worked on a testsuite for a nRF52840-based device some months ago that included measuring idle power consumption, and as far as I remember it still had excessive consumption even after clearing CTRL/STAT, so I ended up having to do a full power cycle anyway.

zyp, 2023-07-29:
Also, this appears to be a duplicate of #1490.

dragonmux, 2023-07-29:
Indeed it is. As that didn't wind up becoming a PR, if the suggested approach on this issue is acceptable for the user, we will implement something and open a PR fixing both issues.