blackmagic-debug/blackmagic #1559: macOS - BDMA with multople probes -n --number option doesn't start server, lists probes instead

chriskuku, 2023-07-22:
macOS (BigSur 11.7.8) but probably not relevant

When starting the BDMA with the --number n or -n n option, when there are multiple probes, 
it doesn't start a server on the selected probe (by sequence number an that one gets e.g. with the -t option) ,
there are still the probes listes and the program exits.

In case there is only one probe, it starts anyway, on that one available probe. Giving a number doesn't have any effect.

dragonmux, 2023-07-22:
This looks to be a regression introduced with PR #1502 during unpicking the logic soup that was `find_debuggers()`. We'll work with Sid to figure out a fix, though this can be worked around effectively by using `-s` and a serial number (or part of a serial number) instead. That will also be more reliable as the enumeration order is not guaranteed (`-n` is supported only for backwards compatibility)

sidprice, 2023-07-22:
This is not a bug but confusion of two BMDA options. 

`-n, --number     Select the target device at the given position in the
                           scan chain (use the -t option to get a scan chain listing)
-P, --probe      Use the <number>th debug probe found while scanning the
                           system, see the output from list for the order`

When multiple probes are attached, use the "-P" option followed by the number in the list output by BMDA.

Example:
`Available Probes:
     Name                 Serial #                  Manufacturer              Version
  1. STLINK-V3            002400423137510A39383538  STMicroelectronics        ---
  2. STM32 STLink         49FF72064986555549112587  STMicroelectronics        ---`

To use the second probe in the list use the command:

`blackmagic -P 2`

Sid

dragonmux, 2023-07-24:
Going to close this as it looks like a case of getting two options mixed and so solved. Please reopen the issue if there's still a problem.

closed 2023-07-24 by dragonmux