blackmagic-debug/blackmagic #1569: Target flash algorithm optimizations

perigoso, 2023-07-28:
This is a list of target implementations that can benefit from a cleanup and optimization of the flash algorithm using the new target flash API implemented in #1205

 - [ ]  ch32f1
 - [ ]  efm32
 - [ ]  lpc_common
 - [ ]  nrf51
 - [ ]  nxpke04
 - [ ]  renesas
 - [ ]  stm32f1
 - [ ]  stm32g0

This list was obtained by manually analyzing the code and is opinionated by nature, may also be incomplete

This issue is meant only to serve as a parent/reference and track the individual cleanups, that will be handled in their own issues/PRs

dragonmux, 2023-08-01:
The STM32F1 Flash erase handling has now been modernised, but we didn't yet touch the Flash write routine. That gets the first target on this list closer to being cleaned up

perigoso, 2023-09-01:
As part of #1399 I'm doing a large rework of ` ch32f1` and ` stm32f1` (maybe ` stm32g0` too at some point), I have not pushed said rework but let this serve as a warning to avoid wasted effort reworking said targets

dragonmux, 2024-11-09:
We did more recently do an overhaul of the ch32f1 support, as part of straightening things out for the native firmware variants introduction, however it can probably do with more work as we didn't address any real structural problems there.

Let us know what you think!