blackmagic-debug/blackmagic #1589: RTT broken on STM32 Blue pill

simonsunnyboy, 2023-08-09:
I tried building the STM32 Blue pill target with current main branch of today.

`$ make PROBE_HOST=swlink ENABLE_RTT=1`

This builds but exceeds the 128K FLASH available and thus cannot be flashed in the end:

` LD      blackmagic.elf
Memory region         Used Size  Region Size  %age Used
             rom:      134788 B       128 KB    102.84%
             ram:       13636 B        20 KB     66.58%
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: address 0x8020ce0 of blackmagic.elf section `.text' is not within region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: blackmagic.elf section `.ARM.exidx' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: address 0x8020ce0 of blackmagic.elf section `.text' is not within region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 3716 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:133: blackmagic.elf] Fehler 1
make: *** [Makefile:17: all] Fehler 2
`

If I omit the ENABLE_RTT flag the build is working however I cannot use RTT then.

sidprice, 2023-08-09:
You can work around this by removing some of the targets from the Makefile, look at the lines following "SRC =" and delete any targets you do not need right now.

dragonmux, 2023-08-09:
We are aware of this problem, as this duplicates issue #1579.

Please note that building with GCC 9 is unsupported now, as is using your distro's toolchain due to issues with miscompilation and you should upgrade your compiler to [ARM's GNU toolchain version 12.2.Rel1](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads#panel3a). The upgraded compiler does much better on code generation and optimisation, which helps alleviate this problem somewhat. More details are available on the linked issue.

simonsunnyboy, 2023-08-10:
> You can work around this by removing some of the targets from the Makefile, look at the lines following "SRC =" and delete any targets you do not need right now.

If that means removing support for NXP or vendors different than ST, this looks like a short-term solution. I will try that.

simonsunnyboy, 2023-08-10:
> We are aware of this problem, as this duplicates issue #1579.
>
> Please note that building with GCC 9 is unsupported now, as is using your distro's toolchain due to issues with miscompilation and you [...]

Update of toolchain is noted but of no concern for me as only the BMP is affected.

I could work around by building the hosted variant which allows me to have working RTT support as a stop-gap even if it not on the probe itself.

closed 2023-08-10 by simonsunnyboy