blackmagic-debug/blackmagic #1590: Enabling extra compile warnings

perigoso, 2023-08-10:
In the process of working on #1528 the new build-system capabilities were leveraged to enable extra warnings.
This issue lists all desired future extra warnings and tracks whether they pass with the current state of the code-base, along with respective cleanups where needed.

Warnings:
- [ ] [arith-conversion](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Warith-conversion)
- [ ] [bad-function-cast](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wbad-function-cast)
- [x] [cast-align=strict](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wcast-align)
- [x] [cast-function-type](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wcast-function-type)
- [x] [cast-qual](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wcast-qual)
- [ ] [conversion](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wconversion)
- [x] [dangling-else](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wdangling-else)
- [ ] [double-promotion](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wdouble-promotion)
- [x] [duplicated-branches](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wduplicated-branches)
- [x] [float-conversion](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wfloat-conversion)
-Wformat=2 -Wformat-overflow=2 -Wformat-signedness -Wformat-truncation
- [ ] [format](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wformat)
	- [format-overflow=2](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wformat-overflow)
	- [format-signedness](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wformat-signedness)
	- [format-truncation](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wformat-truncation)
- [x] [implicit-fallthrough](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wimplicit-fallthrough)
- [x] [maybe-uninitialized](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmaybe-uninitialized)
- [x] [missing-attributes](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-attributes)
- [x] [missing-braces](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-braces)
- [ ] [char-subscripts](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wno-char-subscripts)
- [x] [null-dereference](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wnull-dereference)
- [ ] [packed](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wpacked)
- [ ] [redundant-decls](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wredundant-decls)
- [x] [return-type](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wreturn-type)
- [x] [sequence-point](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wsequence-point)
- [ ] [shadow=local](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wshadow)
- [ ] [sign-conversion](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wsign-conversion)
- [ ] [stack-protector](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstack-protector)
- [x] [strict-aliasing](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstrict-aliasing)
- [x] [strict-overflow=3](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstrict-overflow)
- [x] [string-compare](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstring-compare)
- [x] [stringop-overflow](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstringop-overflow)
- [x] [unknown-pragmas](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunknown-pragmas)
- [x] [unsafe-loop-optimizations](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunsafe-loop-optimizations)
- [x] [unsuffixed-float-constants](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunsuffixed-float-constants)
- [x] [unused-const-variable=2](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused-const-variable)
- [x] [unused-local-typedefs](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused-local-typedefs)
- [x] [unused](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused)
- [x] [vla-parameter](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wvla-parameter)
- [ ] [vla](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wvla)
- [ ] [pedantic](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wpedantic)
        - [ ] [pointer-arith](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wpointer-arith)
	- [ ] empty initializers
	- [ ] range expressions
	- [ ] enum forward ref
	- [ ] void with return
	- [ ] binary constants

dragonmux, 2024-01-14:
Starting to really get close to this being fixed - chipping away at the warnings little by little as things come into scope to look at. Delighted by what we're seeing!

dragonmux, 2024-12-17:
#2026 fixes some more `-Wpedantic` issues and cranks the warning level Meson supplies up to 3 to give us more complete diagnostics enforcement. There are definitely still more warnings that need work done to enable, but the list is nicely dwindling. We're going to make finishing this work a goal for v2.1, so we'll put this tracking issue into that milestone.