blackmagic-debug/blackmagic #1600: blackmagic not working anymore on stlinkv2 clones

franzflasch, 2023-08-12:
Hi!

I would like to report an issue with with bmp on a cheap stlinkv2 clone. The ones you can buy on amazon for ~3$. I encountered that bmp does not work anymore with latest main (commit: f0427959ff30f94a053d614014ee14217eed560a)

First thing I noticed is that the memory with the latest commit is already pretty full:

```
  LD      blackmagic.elf
Memory region         Used Size  Region Size  %age Used
             rom:      129052 B       128 KB     98.46%
             ram:       12152 B        20 KB     59.34%
  OBJCOPY blackmagic.bin
  CC      platforms/stlink/usbdfu.c
  CC      platforms/common/stm32/dfucore.c
  CC      platforms/common/stm32/dfu_f1.c
  LD      blackmagic_dfu.elf
  OBJCOPY blackmagic_dfu.bin
  OBJCOPY blackmagic_dfu.hex
```

I flashed it to the device nonetheless and tried it. It gets recognized just fine in dmesg:

```
[115606.841454] usb 3-4.4.4.4.2: USB disconnect, device number 54
[115609.499749] usb 3-4.4.4.4.2: new full-speed USB device number 55 using xhci_hcd
[115609.732469] usb 3-4.4.4.4.2: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.09
[115609.732475] usb 3-4.4.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[115609.732477] usb 3-4.4.4.4.2: Product: Black Magic Probe (ST-Link/v2) v1.9.0-1119-gf0427959
[115609.732479] usb 3-4.4.4.4.2: Manufacturer: Black Magic Debug
[115609.732480] usb 3-4.4.4.4.2: SerialNumber: E0DDDBC3
[115609.823547] cdc_acm 3-4.4.4.4.2:1.0: ttyACM0: USB ACM device
[115609.829543] cdc_acm 3-4.4.4.4.2:1.2: ttyACM1: USB ACM device
```

However it does not work:

```
#> arm-none-eabi-gdb -ex "target extended-remote /dev/ttyACM0" -ex "monitor swdp_scan"
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyACM0
Target voltage: 3.28V
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

So I decided to test with the latest release branch v.1.9 with latest commit 4e375910ef90356623d18493564168ae7845b8aa

It uses less memory:

```
  LD      blackmagic.elf
Memory region         Used Size  Region Size  %age Used
             rom:      118752 B       128 KB     90.60%
             ram:       11888 B        20 KB     58.05%
  OBJCOPY blackmagic.bin
  CC      platforms/stlink/usbdfu.c
  CC      platforms/stm32/dfucore.c
  CC      platforms/stm32/dfu_f1.c
  LD      blackmagic_dfu.elf
  OBJCOPY blackmagic_dfu.bin
  OBJCOPY blackmagic_dfu.hex
```

This is the dmesg output for v1.9:

```
[115951.766285] usb 3-4.4.4.4.2: new full-speed USB device number 59 using xhci_hcd
[115951.992599] usb 3-4.4.4.4.2: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.00
[115951.992614] usb 3-4.4.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[115951.992616] usb 3-4.4.4.4.2: Product: Black Magic Probe (ST-Link/v2) v1.9.1-1-g4e375910
[115951.992617] usb 3-4.4.4.4.2: Manufacturer: Black Magic Debug
[115951.992618] usb 3-4.4.4.4.2: SerialNumber: E0DDDBC3
[115952.095636] cdc_acm 3-4.4.4.4.2:1.0: ttyACM0: USB ACM device
[115952.101578] cdc_acm 3-4.4.4.4.2:1.2: ttyACM1: USB ACM device
```

And it also works just fine:

```
#> arm-none-eabi-gdb -ex "target extended-remote /dev/ttyACM0" -ex "monitor swdp_scan"
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyACM0
Target voltage: 3.27V
Available Targets:
No. Att Driver
 1      STM32F40x M4
```

So I must assume that between v1.9 and the latest main f0427959ff30f94a053d614014ee14217eed560a some change must have broke bmp on those cheap stlinks.

Let me know if I should try and test anything.

dragonmux, 2023-08-12:
hi franzflasch - a couple of things that would be useful here:

* What compiler are you using (and no, we don't just mean "GCC 12.2" or so, we mean the full version string from `arm-none-eabi-gcc --version` as the exact specifics matter)?
* How did you compile the firmware exactly? (what `make` line did you use

We're well aware about how full Flash is getting and that's expected and normal - the project got big, and the current build system affords us insufficient build flexibility to address this sufficiently but this is on the roadmap to fix.

Given the fault you're seeing, it would be useful if you could also try PR #1584 which fixes a few bugs with how the early phases of ADIv5 bring-up work and is pending review merge.

We specifically support only ARM's own builds of GCC right now due to miscompilation and optimisation issues we've been working on fixing, so if you are not using [ARM's GNU toolchain version 12.2.Rel1](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads#panel3a) then we suggest you grab this and use it for the time being.

franzflasch, 2023-08-12:
Hi dragonmux!

Thanks for the fast answer! I am using my own built toolchain with latest gcc 13.1.0:

```
#> arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 13.1.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

It works just fine with v1.9. But I will try with your suggested compiler. Thanks for the hint.

Also I will try with PR #1584 and will report back. Thanks for the help!

franzflasch, 2023-08-12:
OK, I've now tried with your suggested toolchain. It is exactly the same issue:

```
#> arm-none-eabi-gdb -ex "target extended-remote /dev/ttyACM0" -ex "monitor swdp_scan"          
GNU gdb (GDB) 13.1                                                                                                                                             
Copyright (C) 2023 Free Software Foundation, Inc.                                                                                                              
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>                                                                                  
This is free software: you are free to change and redistribute it.                                                                                             
There is NO WARRANTY, to the extent permitted by law.                                                                                                          
Type "show copying" and "show warranty" for details.                                                                                                           
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".                                                                                
Type "show configuration" for configuration details.                                                                                                           
For bug reporting instructions, please see:                                                                                                                    
<https://www.gnu.org/software/gdb/bugs/>.                                                                                                                      
Find the GDB manual and other documentation resources online at:               
    <http://www.gnu.org/software/gdb/documentation/>.                          
                                                                               
For help, type "help".                                                                                                                                         
Type "apropos word" to search for commands related to "word".                                                                                                  
Remote debugging using /dev/ttyACM0                                                                                                                            
Target voltage: 3.26V                                                                                                                                          
Ignoring packet error, continuing...                                                                                                                           
Ignoring packet error, continuing...                                           
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

However PR #1584 applied it works! :)

```
arm-none-eabi-gdb -ex "target extended-remote /dev/ttyACM0" -ex "monitor swdp_scan"
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi". 
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyACM0
Target voltage: 3.26V
Available Targets:
No. Att Driver
 1      STM32F40x M4
```

With that PR it even works with my own toolchain. Thanks for your help!

dragonmux, 2023-08-12:
We'll link this issue up with that PR then as being solved by it :)

closed 2023-08-20 by esden