blackmagic-debug/blackmagic #1644: blackpill-f4: #define PLATFORM_HAS_POWER_SWITCH leads to error: implicit declaration of function 'adc_start_conversion_direct'

lenvm, 2023-10-16:
Using 9fe5889, adding ```#define PLATFORM_HAS_POWER_SWITCH``` to blackpill-f4.h and compiling using the command ```make PROBE_HOST=blackpill-f401cc``` leads to the following error.

```vim
  CC      platforms/common/stm32/timing_stm32.c
platforms/common/stm32/timing_stm32.c: In function 'sys_tick_handler':
platforms/common/stm32/timing_stm32.c:110:25: error: implicit declaration of function 'adc_start_conversion_direct'; did you mean 'adc_start_conversion_injected'? [-Werror=implicit-function-declaration]
  110 |                         adc_start_conversion_direct(ADC1);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         adc_start_conversion_injected
cc1: all warnings being treated as errors
make[1]: *** [timing_stm32.o] Error 1
make: *** [all] Error 2
```

This error was not present a couple months ago when compiling in the same way.

dragonmux, 2023-10-16:
Ah, that's our bad. PR #1434 wasn't really careful enough to define the tpwr monitoring code only for native or make sure we'd done it in a way that's compatible with all MCUs that make use of that SysTick handler.

Looks like the F4's have different ADC routines and interfaces.. We've got two options here - one is to change the `#ifdef` guard to check not only that `PLATFORM_HAS_POWER_SWITCH` is defined but that the platform is at least targeting a STM32F1; the other is to refactor the ADC aspect of this into the platform code properly and then implement it for the F4 platforms too.

lenvm, 2023-10-17:
Hi @dragonmux, thanks for your quick reply!

Of course, the ideal solution would be to refactor the code properly and to implement the feature for the F4 platforms as well. I would however already be happy if the `#ifdef` was changed as a remedial action, until there is time to do the refactoring.

dragonmux, 2023-10-17:
Before we go and open a PR with an attempted fix then.. could you change the offending `#ifdef` to: `#if defined(PLATFORM_HAS_POWER_SWITCH) && define(STM32F1)` and confirm for us that it does in fact fix the issue?

lenvm, 2023-10-18:
Hi @dragonmux, I have implemented the changes necessary to fix the immediate issue. I have committed the changes and created pull request #1648.

PR #1648 (2023-10-18): Fix: platforms/common/stm32: fix for adc_start_conversion_direct not being defined for STM32F4

## Detailed description
The function `adc_start_conversion_direct()` is only defined for STM32F1, and not for STM32F4.

This caused a compilation issue for STM32F4 based boards.

This pull request fixes this issue, as described in #1644, by adding a check if the platform is based on an STM32F1.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](h [... cut]