blackmagic-debug/blackmagic #1653: Error on STM32U575ZIT6Q MCU Cortex-M33

BolivarTech, 2023-10-21:
On an STM32U575ZIT6Q MCU with the firmware Release v1.9.2 on the BMP I'm get the follow error and the debug stop:

Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x4820
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x4820 M33
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x08001088 in HAL_GPIO_Init (GPIOx=0xd9548, pGPIO_Init=0xffffffff) at ./Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_gpio.c:321
321             tmp &= ~((0x0FUL) << (8U * (position & 0x03U)));
Program stopped, probably due to a reset and/or halt issued by debugger
warning: could not convert '_tx_thread_created_ptr' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.

dragonmux, 2023-10-21:
Please try firmware v1.10.0-rc1. This is a STM32 part and this looks like one that was not supported by v1.9.2 - but is supported in v1.10.0 which is currently undergoing release candidates pending full release.

dragonmux, 2023-10-31:
Please let us know if your issue is solved with the v1.10.0 firmware now it's been released.

BolivarTech, 2023-10-31:
Hello Rachel,

Yes it is.

Thanks.

Julian Bolivar 

⁣Get BlueMail for Android ​

closed 2023-10-31 by dragonmux