blackmagic-debug/blackmagic #1663: samd: timeout when connecting to samc21

koendv, 2023-10-30:
Attached the bmda output when connecting to an atsamc21. 
```blackmagic -tv 5``` was run twice.
The first time there is a 2 second delay; the second time response is instantaneous.
[log.txt](https://github.com/blackmagic-debug/blackmagic/files/13207132/log.txt)

dragonmux, 2023-10-30:
Interesting.. both succeeded, despite the warning about failing to halt the CPU.

Our bad though that we said `-v 5` and should have asked for `-v 29` so we get the full ADIv5 conversation - could you recapture the two traces at this higher verbosity level please?

koendv, 2023-10-30:
OK. ```blackmagic -tv 29```, run twice. Log attached.
[log2.txt](https://github.com/blackmagic-debug/blackmagic/files/13207918/log2.txt)

samd.c has some routines that look like workarounds for errata.. e.g. "Only required for SAM D20 _Revision B_ Silicon". samc21 errata are [here](https://ww1.microchip.com/downloads/en/DeviceDoc/80000740D.pdf).