blackmagic-debug/blackmagic #1697: Segmentation fault when listing probes in hosted mode

Tirpitz93, 2023-12-11:
I am trying to run blackmagic in hosted mode on a raspberry PI 4.

When I run it without probes attached it give 
```
pi@raspberrypi:~/blackmagic/src $ ./blackmagic 
No probes found
```
as expected
but when I run it with the probe attached nothing happens.
```
pi@raspberrypi:~/blackmagic/src $ ./blackmagic 
pi@raspberrypi:
```
When I run with `-l` I get:
```
pi@raspberrypi:~/blackmagic/src $ ./blackmagic -l
Available Probes:
     Name                 Serial #                  Manufacturer              Version
  1. 003800274D4B500D20373831 ---                       STLINK-V3                 (null)
Segmentation fault
```

I tried debugging the app, and I get a segmentation fault :
```

pi@raspberrypi:~/blackmagic/src $ gdb blackmagic 
GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from blackmagic...
(gdb) run -l
Starting program: /home/pi/blackmagic/src/blackmagic -l
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xf7d82440 (LWP 1508)]
Available Probes:
     Name                 Serial #                  Manufacturer              Version
  1. 003800274D4B500D20373831 ---                       STLINK-V3                 (null)

Thread 1 "blackmagic" received signal SIGSEGV, Segmentation fault.
find_debuggers (cl_opts=cl_opts@entry=0x619e0 <cl_opts>, info=info@entry=0x61594 <bmda_probe_info>) at platforms/hosted/bmp_libusb.c:539
539                             DEBUG_WARN(" %2zu. %-20s %-25s %-25s %s\n", position, probe->product, probe->serial, probe->manufacturer,
(gdb) 
```

I built from the current main:
```
make PROBE_HOST=hosted HOSTED_BMP_ONLY=1
```
git status
```
pi@raspberrypi:~/blackmagic $ git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

dragonmux, 2023-12-11:
The underlying platform does not matter in this case - just that you're on *a* Linux platform.
If you've still got that debug session open, please run `print probe` as locals are not mentioned and that's the most likely thing to be `NULL`.

Given you're trying to use an ST-Link v3 (by the looks), you need `HOSTED_BMP_ONLY` to be 0, not 1. As a note for future debugging runs, you can run `gdb` with arguments pre-specified as in `gdb --args ./blackmagic -l` without having to then `run -l` or `args -l` & `run` in GDB.

Tirpitz93, 2023-12-11:
> The underlying platform does not matter in this case - just that you're on _a_ Linux platform. If you've still got that debug session open, please run `print probe` as locals are not mentioned and that's the most likely thing to be `NULL`.
> 
> Given you're trying to use an ST-Link v3 (by the looks), you need `HOSTED_BMP_ONLY` to be 0, not 1. As a note for future debugging runs, you can run `gdb` with arguments pre-specified as in `gdb --args ./blackmagic -l` without having to then `run -l` or `args -l` & `run` in GDB (just `run` and it'll use the options from the CLI).

Thank you for the quick reply,  

After I ran `make clean` and then `make PROBE_HOST=hosted`
I no longer get a seg fault. its probable that my build wasnt clean last time.
and I had run a combination of `HOSTED_BMP_ONLY=` `1` & `0` previously that may well have messed up the build.

```
pi@raspberrypi:~/blackmagic/src $ ./blackmagic 
Trying ST-Link reset
Listening on TCP port: 2000
```

dragonmux, 2023-12-11:
Ah, yes, if you don't fully clean between switching in the current Makefile build system, it puts the build state into this weird in-between state. Glad it's all working now though 🙂 

Closing the issue as solved then - thank you for the feedback! Thankfully, the new Meson build system that's on the way and the planned replacement kbuild-style Makefile one should entirely avoid and solve these problems.