blackmagic-debug/blackmagic #1749: GDB crash when attaching while under GEF

Grazfather, 2024-01-27:
GDB 13.2, arm-none-eabi-gdb on MacOS Sonoma (installed with homebrew)

```
❯ arm-none-eabi-gdb -q
GEF for darwin ready, type `gef' to start, `gef config' to configure
88 commands loaded and 5 functions added for GDB 13.2 in 0.00ms using Python engine 3.11
[=] [remote] initializing remote session with /dev/cu.usbmodem72AE15F41 target 1 under /var/folders/wp/n7lqy3px1zj4frbssdpqwdp40000gp/T/tmpci8scds3
[=] [remote] Installing new objfile handlers
[=] [remote] Executing 'target extended-remote /dev/cu.usbmodem72AE15F41'
[remote] start_remote_1: enter
  [remote] Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+#ec
  [remote] Received Ack
  [remote] Packet received: PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
  [remote] packet_ok: Packet qSupported (supported-packets) is supported
  [remote] Sending packet: $vMustReplyEmpty#3a
  [remote] Received Ack
  [remote] Packet received:
  [remote] Sending packet: $!#21
  [remote] Received Ack
  [remote] Packet received: OK
  [remote] Sending packet: $Hg0#df
  [remote] Received Ack
  [remote] Packet received: OK
  [remote] Sending packet: $qXfer:features:read:target.xml:0,3fb#46
  [remote] Received Ack
  [remote] Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture> <feature name="org.gnu.gdb.arm.m-profile"><reg name="r0" bitsize="32"/><reg name="r1" bitsize="32"/><reg name="r2" bitsize="32"/><reg name="r3" bitsize="32"/><reg name="r4" bitsize="32"/><reg name="r5" bitsize="32"/><reg name="r6" bitsize="32"/><reg name="r7" bitsize="32"/><reg name="r8" bitsize="32"/><reg name="r9" bitsize="32"/><reg name="r10" bitsize="32"/><reg name="r11" bitsize="32"/><reg name="r12 [508 bytes omitted]
  [remote] Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11
  [remote] Junk: qXfer:features:read:target.xml:3fb,3fb
  [remote] Received Ack
  [remote] Packet received: me="no"/></feature></target>
  [remote] Sending packet: $qXfer:features:read:target.xml:416,3fb#b1
  [remote] Received Ack
  [remote] Packet received: l
  [remote] Sending packet: $qTStatus#49
  [remote] Received Ack
  [remote] Packet received:
  [remote] packet_ok: Packet qTStatus (trace-status) is NOT supported
  [remote] Sending packet: $?#3f
  [remote] Received Ack
  [remote] Packet received: W00
[remote] start_remote_1: exit
[=] [remote] Executing 'monitor tpwr enable'
[remote] Sending packet: $qRcmd,7470777220656e61626c65#07
[remote] Received Ack
[remote] Packet received: O456E61626C696E672074617267657420706F7765720A
Enabling target power
[remote] Packet received: OK
[=] [remote] Executing 'attach 1'
[remote] Sending packet: $vAttach;1#37
[remote] Received Ack
[remote] Packet received: T05thread:1;
[remote] packet_ok: Packet vAttach (attach) is supported
[remote] Sending packet: $qC#b4
[remote] Received Ack
[remote] Packet received: QC1
../../gdb/thread.c:85: internal-error: inferior_thread: Assertion `current_thread_ != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
----- Backtrace -----
error creating backtrace: executable file is not an executable
---------------------

This is a bug, please report it.  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.

Abort trap: 6
```

dragonmux, 2024-01-27:
Having poked at this a bit with you in Discord, it appears to be an interaction between GEF, GDB and BMD around when GDB's getting details about the newly attached target's threads environment.

Further debugging is required to figure out what exactly and who's bug this is quite (beyond GDB still refusing to properly fix #929 which precipitates the entire problem to begin with)

Grazfather, 2024-01-27:
Easy repro:
gdbinit:
```
source hook_continue.py
target extended-remote /dev/cu.usbmodem72AE15F41
monitor tpwr enable
monitor swdp_scan
attach 1
```

hook_continue.py:
```python
def f(_):
    gdb.selected_frame()

gdb.events.cont.connect(f)
```

Grazfather, 2024-01-27:
Seems that the continue event is fired out _while attaching_. While this is happening the `current_thread_` is set to `nullptr`, so `gdb.selected_frame()` cannot be called.

dragonmux, 2024-01-27:
We can confirm with this trivial repro that yep, crashes!

Grazfather, 2024-01-28:
This clearly isn't a BMP bug, but I haven't been able to reproduce when attaching to a local process... It seems that the hooks are not called in that flow, which makes sense since we see `attach_command` call `extended_remote_target::attach`.

I've filed a bug with gdb: https://sourceware.org/bugzilla/show_bug.cgi?id=31303

closed 2024-11-24 by Grazfather