blackmagic-debug/blackmagic #1768: Meson build including more targets than configured.

sidprice, 2024-02-20:
It appears the Meson build system is not following the configured target set. On a Windows PC I tried to build the Native platform in order to make use of RTT. The initial build failed due to the flash being over-filled.

To work around this the following Meson configure command was used:

`meson configure "-Dtargets=cortexm,stm"`

However, when the build was run it was noted the flash was still 97% used.

Investigation showed more targets than this configured had been built. For example, the SAMD range was included.

perigoso, 2024-02-23:
I believe this might be what I hit while working on the initial PR, see [this review comment](https://github.com/blackmagic-debug/blackmagic/pull/1528#discussion_r1448958782_).

It's likely a meson bug, but I have have not tracked it down yet. @sidprice could you provide some details about your system so we can maybe pin this down?

- What os/distribution are you using?
- What meson version did you use (`meson --version`)?
- How did you install it?

In the meantime, using the `setup` command with the `--reconfigure` flag should work as a workaround:
`meson setup --reconfigure "-Dtargets=cortexm,stm"`

sidprice, 2024-02-23:
What os/distribution are you using? <-- Windows 11
What meson version did you use <-- 1.0.0
How did you install it? <-- Using MSYS2

And thank you for the reconfigure tip, that worked for me :)

dragonmux, 2024-12-16:
This has been fixed by documentation in #2019 - hopefully this fix suffices. If it does, we'll mark this issue as fixed by that PR.