blackmagic-debug/blackmagic #1800: Can't compile bmp for stlinkv2 target

tixlegeek, 2024-04-17:
Hi,
I'm trying to update the formware of a stlinkv2 clone as the README says, on Debian11. 

### Issue:

The linking fails with this error message:
`/opt/arm/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: address 0x8020820 of blackmagic.elf section `.text' is not within region `rom'
/opt/arm/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: blackmagic.elf section `.data' will not fit in region `rom'
/opt/arm/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: address 0x8020820 of blackmagic.elf section `.text' is not within region `rom'
/opt/arm/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 2436 bytes
Memory region         Used Size  Region Size  %age Used
             rom:      133508 B       128 KB    101.86%
             ram:       12140 B        20 KB     59.28%
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:170 : blackmagic.elf] Erreur 1
`
### Reproduce:

0.try 1: install debian's official repo version of none-eabi-arm-gcc [ERROR can't link]
0.try 2: install ARM's official toolchain version  8 of the toolchain [ERROR can't link]
0.try 3: install ARM's official toolchain version 12.2-rel1 [ERROR can't link]

1. Recursively clone the bmp repo.
2. make clean
3.  `make PROBE_HOST=stlink ST_BOOTLOADER=1`

I tryed multiple none-eabi toolchains as mentioned in multiple topics on the subject, those 102% of rom are the best i got.

Can someone help me with this?

Thanks for your great work, and have a nice day.

dragonmux, 2024-04-17:
This is a known issue. Please either use the Meson build system and disable a target or two you're not planning to use, or edit `src/Makefile` to disable a target or two by *deleting* a target's .c from the main `SRC +=` block.

There's nothing in particular the project can do about this as we're just simply reaching the limits of the Flash on these devices, and keeping a complete list of all the permutations and combinations of probes the firwmare has been ported to completely updated with what targets need disabling to make it fit (and hoping those aren't the ones the user wants(!)) is not something we've got the bandwidth to do. The Meson option is by far the best on this score because you can so easily customise what set of targets you want your build to support with `-Dtargets=` and chip/chop + change that trivially.

tixlegeek, 2024-04-17:
well, that worked perfectly fine. Thank's a lot. I understand the limitations. Maybe it would be a good idea to add a note to the readme about that? I guess a lot of people could fancy this tip while using these cheap debuggers.

dragonmux, 2024-04-17:
Yes, having a note in the README would be a good idea - you'd be welcome to contribute that if you'd like, or it'll go on our pile to fix when we get a moment to address it. Happy to hear you're up and working now though.

tixlegeek, 2024-04-17:
It'll be my pleasure. I'lldo this tonight, after work :smile: