blackmagic-debug/blackmagic #1836: Unable to recover from loss-of-target without `abort()`

xobs, 2024-05-29:
It is not possible to recover from the loss of a target.

When a target is lost, an exception is raised. The target list is then freed:

https://github.com/blackmagic-debug/blackmagic/blob/e4da644bb8c0c81498a0f3819bbd99fd6627cd8e/src/main.c#L87-L92

The problem comes from the fact that `target_list_free()` is not contained within a `try {}` block. If there is still a target attached it will attempt to do various things such as resetting breakpoints on the target which no longer exists which will raise another exception and cause a reboot.

dragonmux, 2024-08-13:
After #1851 was merged to `main`, this should have been resolved we believe - could you please test again and let us know if it is or not?

xobs, 2024-08-29:
I've finally managed to test this, and it does indeed work now. Thanks for the fix!

closed 2024-08-29 by xobs