blackmagic-debug/blackmagic #1840: Debug app listens only to IPv6

hi0t, 2024-06-04:
I built a bmda app  v1.10.2 for Windows. Launched the app:
```
blackmagic.exe
Listening on TCP port: 2000
``` 
Now I'm trying to connect via telnet:
```
telnet 192.168.8.64 2000
Connecting To 192.168.8.64...Could not open connection to the host, on port 2000: Connect failed
```
But on IPv6 it works:
```
telnet fe80::ae87:ce3d:8e4b:9d49%14 2000
```
Typically, network applications listen to both protocols. I see no reason to limit them to only the IPv6 protocol.

Also in the help it is indicated that only the localhost is listened to:
```
Usage: blackmagic.exe [-h | -l | [-v BITMASK] [-O] [-d PATH | -P NUMBER | -s SERIAL | -c TYPE]
        [-n NUMBER] [-j | -A] [-C] [-t | -T] [-e] [-p] [-R[h]] [-H] [-M STRING ...]
        [-f | -m] [-E | -w | -V | -r] [-a ADDR] [-S number] [file]]

The default is to start a debug server at localhost:2000
```
But in fact, all addresses are listened to. Probably worth fixing the help?

dragonmux, 2024-06-04:
This is solved on `main` - please give it a look. `localhost` even on Windows refers to 127.0.0.1, for what it's worth - ie, the loopback interface; unsure why Windows is allowing the connection from the IPv6 link local address for your NIC, though it might well be some transparent address rewriting in the networking stack making that work.

hi0t, 2024-06-04:
Yes, you are absolutely right. Rebuilt from the main branch, now the utility listens to both protocols. Thank you for such prompt support!

closed 2024-06-04 by hi0t

TechnoMancer, 2024-06-07:
This does sound like a manifestation of pre #1776 behaviour on windows, where it would get the IPv6 address from getaddrinfo but on windows the IPV6_ONLY socket option is set by default so the socket would only accept v6 connections.

It would probably make sense to fix the usage info and BMDA readme to say it binds to the any address not specifically localhost too.

dragonmux, 2024-06-07:
The documentation correction sounds like a reasonable suggestion 👍🏼