blackmagic-debug/blackmagic #1844: samd: regression in `monitor swdp_scan`

maksimdrachov, 2024-06-17:
I'm using a `samc21g`, support for which has been introduced in [this pull request](https://github.com/blackmagic-debug/blackmagic/pull/1659).

It works when I flash that particular commit, however it fails to recognize the MCU when I'm using v1.10.2.

Not recognizing:

[image]

Recognizing:

[image]

dragonmux, 2024-06-17:
Please give latest `main` a try. It's entirely likely that the bug's already been fixed pending the v2.0 release.

maksimdrachov, 2024-06-18:
Hi,

I've re-tried with the latest commit [aa67538](https://github.com/blackmagic-debug/blackmagic/commit/aa6753833764b41697900536395bb343f611377d) and it looks like you're right.

Thanks.

[image]

closed 2024-06-18 by maksimdrachov