blackmagic-debug/blackmagic #1845: Semihost: regression from v1.9.3 to v1.10

maksimdrachov, 2024-06-18:
One of my semihost on-target tests is failing when updating the BlackMagicProbe firmware from v1.9.3 to the latest aa6753833764b41697900536395bb343f611377d in main.

The issue arises when I try to write a negative value through semihost:

```cpp
auto val = semihost_persistence::add("./logs/test_add", -3);
zassert_equal(-3, val.value());
```

(`semihost_persistence::add` will rely on both read as well as write calls)

Positive values work in both versions.

This is using SAMC21.

pavel-kirienko, 2024-06-18:
The problem is also found in the v1.10.2 release.

ALTracer, 2024-06-18:
Weird. These are semihosting file read & write calls? Please exit TUI, enable gdb logging for remote and record to file something more detailed than a word-wrapped screenshot.
Relevant merged PRs are 1686, but also 1726, 1730, if it's pressing you can try to manually test or bisect across them.

dragonmux, 2024-06-18:
> The problem is also found in the v1.10.2 release.

Semihosting is known slightly broken in that release, Pavel, and is nothing to do with this. See #1732 for the complete low-down on that. As this user is testing `main`, this is irrelevant to them as they're testing post-1732.

dragonmux, 2024-06-18:
> The issue arises when I try to write a negative value through semihost:
>
> ```c++
> auto val = semihost_persistence::add("./logs/test_add", -3);
> zassert_equal(-3, val.value());
> ```
>
> (semihost_persistence::add will rely on both read as well as write calls)

It would be helpful if you can share what your semihost_persistence class instance is doing please and which semihosting call it's trying to make when it appears to go wrong. Likewise, please drop out the TUI and provide: `bt` + `info regs` in your `semihost_exec()`. There's just too little information in your screenshot to go on and without also seeing the output of semihosting that's wrong, there's nothing here to debug right now.

maksimdrachov, 2024-06-19:
After re-testing (note for all three below, I've applied a patch that added [samc21 support](https://github.com/blackmagic-debug/blackmagic/commit/f1cd9f2bd36c29e14ceee6bf3e0f6a4e42552ca8)).
I have concluded the following:

- v1.9.3 works.
[gdb_v1.9.3.txt](https://github.com/user-attachments/files/15899050/gdb_v1.9.3.txt)
- v1.10.0 semihost doesn't work at all, I will not consider this for now.
- v1.10.2 semihost works, but not for negative values
[gdb_v1.10.2.txt](https://github.com/user-attachments/files/15898925/gdb_v1.10.2.txt)
- main semihost works, but not for negative values
[gdb_main.txt](https://github.com/user-attachments/files/15899168/gdb_main.txt)

(All compiled using `gcc-arm-none-eabi-9-2020-q2-update`, v1.9.x don't compile when using more recent versions)

---

Here's an attempt at debugging, using the last main version:

[gdb_debug.txt](https://github.com/user-attachments/files/15899369/gdb_debug.txt)

> Likewise, please drop out the TUI and provide: bt + info regs in your semihost_exec().

I'm assuming you mean when the bug occurs? The issue is that when it does, the gdb session hangs and I'm not able to put in any command anymore:

[image]

Todo after lunch:
- [ ] _It would be helpful if you can share what your semihost_persistence class instance is doing in the add method please and which semihosting call it's trying to make when it appears to go wrong._
- [ ] _I'm not sure whether your probes support ENABLE_DEBUG=1 or whether BMDA would reproduce it._
- [ ] _Likewise as ALTracer suggests, use BMDA as blackmagic -v 5 to get a log of the semihosting activity and possibly even the hang. We can't even tell which syscall is being made to know where to start looking._

maksimdrachov, 2024-06-19:
I hope this is helpful for debugging:

- working, with semihost logging `3`:
[gdb_main_3.txt](https://github.com/user-attachments/files/15900639/gdb_main_3.txt)
- not working, with semihost logging `-3`:
[gdb_main_min3.txt](https://github.com/user-attachments/files/15900645/gdb_main_min3.txt)

I think the cause of the issue is to be found here:

[image]

ALTracer, 2024-06-19:
I've found the problem on https://github.com/blackmagic-debug/blackmagic/blob/aa6753833764b41697900536395bb343f611377d/src/target/semihosting.c#L180-L190
In PR1686 I fixed [whatever existed previously] incompletely, and the 'F' check also matches against m-replies containing first (LE) byte of 0xF0..0xFF emitted by BMP, because a) BMP reuses the gdb_packet_buffer due to SRAM constraints; b) no extra info is used otherwise.
I reproduced this on Koen's sketch `SemihostingTest.ino` modified, and `blackmagic-test-fw-archive` does not contain tests for file read/writes.

closed 2024-06-20 by dragonmux