blackmagic-debug/blackmagic #1883: stepping on empty loop

lzace817, 2024-07-30:
whenever I step on a empty loop, gdb hangs and `Ctrl-c` stop working until I reset the board.

## Steps to reproduce
- setup the code bellow 
``` C
#include <stdio.h>
#define SEMIHOSTING
#ifdef SEMIHOSTING
void initialise_monitor_handles(void);
#endif //SEMIHOSTING
volatile int s;
int main(void)
{
#ifdef SEMIHOSTING
    initialise_monitor_handles();
#endif // SEMIHOSTING
    printf("Hello, World!\n");
    while(1) {
    }
}
```

- set a breakpoint just before the loop
- `next` and `next`

I have this logs from a rust test, but the effective setup is pretty much the same.

``` bash
$ ./blackmagic -v 1
Black Magic Debug App (for BMP only) v1.10.2
Using:
 _v1.10.2 BlackPill-F401CC 306D35993235
Listening on TCP port: 2000
Got connection
Speed set to 7.000MHz for SWD
Switching out of dormant state into SWD
DP DPIDR 0x1ba01477 (v1 rev0) designer 0x43b partno 0xba
AP   0: IDR=14770011 CFG=00000000 BASE=e00ff003 CSW=a3000040 (AHB3-AP var1 rev1)
Halt via DHCSR(01030003): success after 14ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 020 Partno 410
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004001bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x411fc231 (M3 var 1 rev 1)
 1 0xe0001000: 0x00000000 <- does not match preamble (0xb105000d)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
 3 0xe0000000: 0x00000000 <- does not match preamble (0xb105000d)
 4 0xe0040000: 0x00000000 <- does not match preamble (0xb105000d)
5 Entry 0xfff42002 -> Not present
ROM: Table END
syscall     SYS_OPEN (800172c 4 3 800172c)
syscall    SYS_WRITE (3 8001670 e 3)
Hello, world!
```

``` bash
$ arm-none-eabi-gdb -q hello.elf
...
(gdb) n
19	    loop {}
(gdb) n
^C^C^C^C^C^C^C^C
cortex_m_rt::Reset () at src/lib.rs:497
497	pub unsafe extern "C" fn Reset() -> ! {
(gdb) 
518	    __pre_init();
(gdb) 
```
 
## Expected behavior
the second `next` should block and run the loop, but `Ctrl-c` should be able to interrupt.
``` bash
Breakpoint 1, main () at dbg-loop.c:4
4	{
(gdb) n
5	    printf("Hello, World!\n");
(gdb) n
Hello, World!
6	    while(1){
(gdb) n
^C
Program received signal SIGINT, Interrupt.
main () at dbg-loop.c:6
6	    while(1){
(gdb) 
```

## Notes
- This also happening with the probe alone.
- Tested target with `-O0` and `-Og`
- Target is well tested and know to be working

dragonmux, 2024-07-30:
Noting that you're using v1.10.2 which has some known issues with semihosting, please can you give latest `main` a try and let us know if things are still unhappy? You can download a [nightly build](https://nightly.link/blackmagic-debug/blackmagic/workflows/build-and-upload/main) to hit the ground running as this does not require a probe firmware change to test.

lzace817, 2024-07-30:
I wasn't able to compile both working firmware and BMDA on my setup.

dragonmux, 2024-07-30:
As said, you do not need to change firmware to test this. You *only* need BMDA. Working firmware, however, can be built with ARM GCC 12.2.Rel1, and a build of the firmware for every supported probe is available from the nightly build downloads too.

lzace817, 2024-07-30:
udev is required? Can't find the probe

lzace817, 2024-07-30:
``` bash
$ ./blackmagic-bmda -d /dev/ttyACM0 -v 1
Black Magic Debug App 763e057
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Setting V6ONLY to off for dual stack listening.
Listening on TCP port: 2000
Got connection
Speed set to 7.000MHz for SWD
Switching out of dormant state into SWD
DP DPIDR 0x1ba01477 (v1 rev1) designer 0x43b partno 0xba
AP   0: IDR=14770011 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var1 rev1)
Halt via DHCSR(01030003): success after 13ms
ROM: Table BASE=0xe00ff000 SYSMEM=1, Manufacturer 020 Partno 410 (PIDR = 0x00000000000a0410)
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004001bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x411fc231 (M3 var 1 rev 1)
 1 0xe0001000: 0x00000000 <- does not match preamble (0xb105000d)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
 3 0xe0000000: 0x00000000 <- does not match preamble (0xb105000d)
 4 0xe0040000: 0x00000000 <- does not match preamble (0xb105000d)
5 Entry 0xfff42002 -> Not present
ROM: Table END
syscall     SYS_OPEN (800172c 4 3 800172c)
syscall    SYS_WRITE (2 8001670 e 2)
```

``` bash
$ arm-none-eabi-gdb -q hello.elf
...
13	    hprintln!("Hello, world!").unwrap();
(gdb) n
Hello, world!
19	    loop {}
(gdb) n
^C^C^C^C^C^C^C^C^C^C^C

^C^C^C^C^C^C^C^C       # PRESSED RESET BUTTON HERE
0xfffece3c in ?? ()
(gdb) 
Cannot find bounds of current function
(gdb) bt
#0  0xfffece3c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

dragonmux, 2024-07-30:
> udev is required? Can't find the probe, trying with -d.

Yes, it always has been for full BMDA builds (ie, non-BMP-only), which are also the only supported configuration for Meson-based builds. Full BMDA uses libusb to find probes and their information (which is also much more reliable than the method BMP-only builds use). Doing this is what makes it require the rules, but then those rules also properly set up the /dev nodes and give you /dev/ttyBmpGdb* and /dev/ttyBmpTarg* which are stable, where /dev/ttyACM* are not as they're affected by enumeration order and other devices on the system.

dragonmux, 2024-07-30:
Just been reviewing the initial post some more and something catches our eye that makes us question the accuracy of the C code translation:

```
$ arm-none-eabi-gdb -q hello.elf
...
(gdb) n
19	    loop {}
(gdb) n
^C^C^C^C^C^C^C^C
cortex_m_rt::Reset () at src/lib.rs:497
497	pub unsafe extern "C" fn Reset() -> ! {
(gdb) 
518	    __pre_init();
(gdb) 
```

This appears to be Rust code being compiled and run? It is possible that `loop {}` does not compile to `while (true) {}` but rather a WFI/WFE instruction. Please test our WDT handling branch (#1882) as this fixes the handling of these instructions on the STM32F1 family, among other parts. Either that, or provide a disassembly of that sequence showing what the compiler is lowering the code to, please.

It would also be useful to investigate why you land back at your reset handler, which indicates the CPU might be taking an exception that results in it rebooting. This is something that can be explored with backtraces (`bt`) and core register dumps (`info reg`). However, this is something to explore only after checking that you're not getting punked by a WFI/WFE instruction making BMD loose the target.

We would suggest running BMDA with `-v 5` instead of `-v 1` to get better logging as this will enable target-level debugging output as well as info-level.

lzace817, 2024-07-30:
rust was not relevant.
semihosting was not relevant.
the board reset because after the lockup, I pressed the reset button on the PCB. There is a coment on the gdb log indicating this moment.

## sample code
``` C
void main(void)
{
    volatile int x;
    x = 0; # set breakpoint here
    while(1) {}
}
```

## server
``` bash
blackmagic-bmda -d /dev/ttyACM0 -v 5
Black Magic Debug App 763e057
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Setting V6ONLY to off for dual stack listening.
Listening on TCP port: 2000
Got connection
Speed set to 7.000MHz for SWD
Switching out of dormant state into SWD
DP DPIDR 0x1ba01477 (v1 rev1) designer 0x43b partno 0xba
AP   0: IDR=14770011 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var1 rev1)
Halt via DHCSR(01030003): success after 12ms
ROM: Table BASE=0xe00ff000 SYSMEM=1, Manufacturer 020 Partno 410 (PIDR = 0x00000000000a0410)
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004001bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x411fc231 (M3 var 1 rev 1)
Calling stm32f1_probe
 1 0xe0001000: 0x00000000 <- does not match preamble (0xb105000d)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
 3 0xe0000000: 0x00000000 <- does not match preamble (0xb105000d)
 4 0xe0040000: 0x00000000 <- does not match preamble (0xb105000d)
5 Entry 0xfff42002 -> Not present
ROM: Table END
stm32f1_flash_erase: at 08000000
stm32f1_flash_write: at 08000000 for 1024 bytes
Speed set to 7.000MHz for SWD
Switching out of dormant state into SWD
DP DPIDR 0x1ba01477 (v1 rev1) designer 0x43b partno 0xba
AP   0: IDR=14770011 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var1 rev1)
Halt via DHCSR(00030003): success after 18ms
ROM: Table BASE=0xe00ff000 SYSMEM=1, Manufacturer 020 Partno 410 (PIDR = 0x00000000000a0410)
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004001bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x411fc231 (M3 var 1 rev 1)
Calling stm32f1_probe
 1 0xe0001000: 0x00000000 <- does not match preamble (0xb105000d)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
 3 0xe0000000: 0x00000000 <- does not match preamble (0xb105000d)
 4 0xe0040000: 0x00000000 <- does not match preamble (0xb105000d)
5 Entry 0xfff42002 -> Not present
ROM: Table END

```

## GDB
``` bash
$ ./empty-loop.sh 
Reading symbols from empty-loop.elf...
Remote debugging using :2000
Target voltage: 
Available Targets:
No. Att Driver
 1      STM32F1 medium density M3
Attaching to program: xxxxxxxxxxxxxxxxxxxxxxxxx/empty-loop.elf, Remote target
0x0800015c in main () at empty-loop.c:5
[... 107 lines cut ...]
 80001aa:	f843 0b04 	str.w	r0, [r3], #4
 80001ae:	e7dd      	b.n	800016c <reset_handler+0x8>
 80001b0:	f843 1b04 	str.w	r1, [r3], #4
 80001b4:	e7de      	b.n	8000174 <reset_handler+0x10>
 80001b6:	f854 3b04 	ldr.w	r3, [r4], #4
 80001ba:	4798      	blx	r3
 80001bc:	e7e6      	b.n	800018c <reset_handler+0x28>
 80001be:	f854 3b04 	ldr.w	r3, [r4], #4
 80001c2:	4798      	blx	r3
 80001c4:	e7e6      	b.n	8000194 <reset_handler+0x30>
 80001c6:	f854 3b04 	ldr.w	r3, [r4], #4
 80001ca:	4798      	blx	r3
 80001cc:	e7e8      	b.n	80001a0 <reset_handler+0x3c>
 80001ce:	bf00      	nop
 80001d0:	080001f8 	.word	0x080001f8
 80001d4:	20000000 	.word	0x20000000
 80001d8:	20000000 	.word	0x20000000
 80001dc:	20000000 	.word	0x20000000
 80001e0:	080001f8 	.word	0x080001f8
 80001e4:	080001f8 	.word	0x080001f8
 80001e8:	080001f8 	.word	0x080001f8
 80001ec:	080001f8 	.word	0x080001f8
 80001f0:	080001f8 	.word	0x080001f8
 80001f4:	080001f8 	.word	0x080001f8
```

dragonmux, 2024-07-30:
Thank you for the extra information, we'll do our best to replicate it here on a spare STM32F103 and get back to you now we're able to be sure it's definitely not a WFI/WFE issue or something else like that.

We can also note from the logs that, unless it's a clone that is not detected properly, the STM32 you've got there is genuine. The BMD scan output would say "(clone)" or the specific kind of clone device such as "GD32F1" or similar if it was a detected clone.

mean00, 2024-08-06:
My 0.02 euro$
This is a gotcha due to how gdb interpret the 'n' command i.e. not a bmp issue.
I've ran into similar issues with other debuggers.
When you do next, gdb does step step step until it reaches the "next line" whatever/wherever it is
In the case of an empty loop, that next line is never reached, because you keep executing the same instruction at the same address, so gdb keeps on stepping forever
( the underlying asm code is something along   "b *" )

There are 2 ways to get out of this : 
- You use the "si" command to  step a single instruction
- you add a __asm__("nop")  inside the loop so that there is at least a different instruction/line

There is a ticket there : https://bugs.launchpad.net/gcc-arm-embedded/+bug/1401565 using a jlink debugger
Same cause, same consequence

dragonmux, 2024-08-06:
Thank you for the reminder to get back to this - and sorry it's been a while since the last reply. We did some more digging and mean00 is indeed correct that this is a compiler and GDB limitation.

Specifically, because the compiler is generating an empty loop, it is unable to generate a debug line entry for the loop body to have the debugger break on. GDB is then unaware of there being any instruction within the loop to breakpoint, and so calculates the next instruction for the breakpoint as the instruction after the loop.

Adding to that, it then causes GDB to exercise a bug whereby because of the lack of line information within the loop, it gets itself stuck waiting for the target to move to any other instruction than the branch-to-self to consider the target halted again - resulting in the behaviour you see. Unfortunately, in this instance, BMD is doing exactly as GDB instructs it, and the bug is in GDB and is caused by a compiler limitation.