blackmagic-debug/blackmagic #1887: PyCortexMDebug with Black Magic Probe: error reading

maksimdrachov, 2024-08-07:
Hi,

There's this handy library called [PyCortexMDebug](https://github.com/blackmagic-debug/blackmagic), however I can't get it to work with Black Magic Probe, if anyone has an idea on how to set it up properly?

[image]

Initially I relied on an Atmel debugger, this was the setup procedure I used: 
https://forum.zubax.com/t/using-pycortexmdebug-with-zephyr-rtos/2174

(I suspect I might need to update something in step 3, in order to make it work?)

dragonmux, 2024-08-07:
From a quick look through, it's a set of GDB Python scripts and https://github.com/bnahill/PyCortexMDebug/blob/master/scripts/gdb.py is the canonical how to set up using this thing..

Contributor litui also has a fork if the Flameeyes one doesn't work out: https://github.com/litui/PyCortexMDebug

Unfortunately, as it looks to be doing everything legit through GDB and using `gdb.selected_inferior().write_memory()` + `gdb.selected_inferior().read_memory()` to access memory, we are reasonably sure this isn't something that BMD is doing wrong as such or that the scripts are doing wrong (it reads like a target access issue or so).

The only way to really know and understand what's going on there would be to get logs of the accesses being made - either via GDB's `set debug remote on`, or by using BMDA with logging enabled (eg `blackmagic -v 5`) - to understand what requests are actually being made and how they're being responded to. Using BMDA to get logs would be the most useful to us to see if this is a BMD bug, a target issue, or a SVD issue.

maksimdrachov, 2024-08-07:
Thank you for the quick reply.

I'll have another look at this when I have a bit more time (currently facing a deadline).

maksimdrachov, 2024-08-22:
For those that stumble on this issue, behold the solution:

`set mem inaccessible-by-default off`

[image]

_(Screenshot for illustration purposes, I've added it to `.gdbinit`, so I don't have to think about it)_

---

As I understand it, the reason this happens is that by default `gdb` will not allow you to examine (`x`) memory outside the ranges specified in the linker file.

closed 2024-08-22 by maksimdrachov

dragonmux, 2024-08-22:
> default `gdb` will not allow you to examine (`x`) memory outside the ranges specified in the linker file.

Slight correction: The regions it uses to determine this are the ones provided by the GDB server (`info mem`) not your linker script. GDB has no idea what is or isn't in your linker script by the time it runs. It does, however, have access the memory mappings the debug server provides saying what is and isn't accessible memory on the target.

maksimdrachov, 2024-08-22:
Interesting!

So I thought those memory areas were stored in the ELF file which I'm passing to `gdb`, but turns out: it's the MCU itself which provides this info.

[image]

dragonmux, 2024-08-22:
Close, it's from the debug server not the MCU - BMD derives the information from the MCU but it's up to the target support code in BMD to define the mappings which are then delivered on attach as part of the GDB remote protocol command `qXfer:memory-map:read::`.