blackmagic-debug/blackmagic #1935: Unknown Device Error with TM4C1230E6PMI

EcoPhDG, 2024-09-18:
Hi,

I just got a new Black Magic Probe and am trying to connect it to my TM4C1230E6PMI. The connection over the JTAG works fine, which I see from the target being powered by the probe. However when I start the profiling utility bmprofile.exe, i get this error Message:

`Please report unknown device with Designer 0x43b Part ID 0x4c4`

[image]

What am I doing wrong here?

dragonmux, 2024-09-18:
Hello, what version of the firmware are you running on your probe please? (`monitor version` in GDB) This is likely a bug fixed in either v1.10.2 or the [current nightlies](https://nightly.link/blackmagic-debug/blackmagic/workflows/build-and-upload/main) so please check that first.

NB: You need to be on a current version of that utility as well if you're going to use the compuphase tooling as otherwise SWO won't make it back to it due to an endpoint change that used to be hard-coded into the tool. Even then, if you're on HW6 (BMP v2.3+) you must run PR #1934 which entirely overhauls and fixes SWO handling to correct for the GD32F103's timers, and this will also break using that utility.

On a positive note, however, [orbuculum](https://github.com/orbcode/orbuculum) has no issues working with the PR firmware and will work out the box with just `./orbuculum -m 250` to start up a trace server listening to the output from the SWO capture engine in the firmware. This will then work with much more than just ITM stimulus and the orbuculum suite contains many more tools for profiling and tracing firmware, making it our preferred package.

EcoPhDG, 2024-09-18:
Thanks for your reply!
My firmware is
`Black Magic Probe v1.10.2, Hardware Version 6`

I will try with the suggested methods and report back once i know more.

dragonmux, 2024-09-18:
Ok, then yeah, you'll have to use the PR linked as that hardware won't work with SWO otherwise. Fixing that has been our focus the last few weeks.