blackmagic-debug/blackmagic #1965: SWD scan fail on RaspberryPi Pico W

ENAC-Robotique, 2024-10-16:
I have trouble debugging on a RaspberryPi Pico WH.
Its working fine on a RaspberryPi Pico.

Anyone having the same issue ?

When I run an SWD can on a pico it returns:
```
No. Att Driver
 1      RP2040 M0+
 2      RP2040 M0+
 3      RP2040 Rescue (Attach to reset) 
 ```

but on a pico W(H):
```
SWD scan failed!
Failed
```

In both cases I connected GND, SDWCLK, SWDIO and reset (RUN).

My probe is a custom one based on the native design (using STM32F103), that I use daily with STM32F4, F7,H7.
I updated it with the latest master.

desertsagebrush, 2024-10-17:
Just to chime in, I have no issues with main for debugging any of the picos (normal, W), though I haven't played with my pico 2 yet. I just retested with the two pico Ws that I have on my desk with a native probe running `v1.10.0-1318-ge7957530`, compiled with arm-provided GCC 12.2.rel1 as suggested by the project, and was able to see/connect/load/debug/etc.

dragonmux, 2024-10-17:
You are indeed completely correct in your final paragraph desertsagebrush! To the letter 😂 - until we can see why you're winding up with a scan failure it's very hard to predict what's wrong and why, but BMDA tells all for this sort of issue.

ENAC-Robotique, 2024-10-17:
Problem solved!
Thank you very much for your time.
I did everything you said, it was not giving a lot of information, and in the end, before quitting, I tried the dumbest thing of all: switching SWDIO and SWDCLK. It worked :rage: .
It seems that all documentation about the picoW is wrong.
On my board, SWDCLK is on the right, SWDIO is on the left.
Exactly the opposite from what is said on the official documentation
https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#raspberry-pi-pico-w-and-pico-wh

closed 2024-10-17 by ENAC-Robotique

dragonmux, 2024-10-17:
Glad to hear you got it figured out - the lack of scan information from BMDA is a solid indicator for such a swap of the two wires as well. Thankfully, RPi Foundation are generally pretty good about fixing things like this if you ping them a message some way or another and let them know what's wrong.