blackmagic-debug/blackmagic #1985: Error writing data to flash (from target-download) - rp2040

0x3333, 2024-11-01:
I'm using a F411CE probe with a rp2040 target in VSCode with Cortex-Debug extension.

When I first try to 'launch' the debugger, I get the error:

`Failed to launch GDB: Error writing data to flash (from target-download)`

When I try again, it works as expected. I notice that the first run, the board is running, application wise. After the error, the CPU halts. That's why the second attempt works.

When the Program stops, when it works is in `frame={addr="0xfffffffe",func="<signal handler called>"}`, when it fails is in `frame={addr="0x10008e16",func="prvIdleTask"....`, which is FreeRTOS.

Am I missing something here?

Here is the Cortex-Debug full debug output with the error(Focus on last 30 lines or so):

```
0000000002+00002: Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
0000000002+00000: "configuration": {
    "name": "BMP Launch",
    "cwd": "/Users/0x3333/Code/Home/g6",
    "executable": "/Users/0x3333/Code/Home/g6/build/gateway.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "bmp",
    "interface": "swd",
    "device": "rp2040",
    "numberOfProcessors": 2,
    "BMPGDBSerialPort": "/dev/cu.usbmodem316F333331321",
    "runToEntryPoint": "main",
    "preLaunchCommands": [
        "set mem inaccessible-by-default off"
    ],
    "showDevDebugOutput": "raw",
    "showDevDebugTimestamps": true,
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "powerOverBMP": "lastState",
    "targetId": 1,
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/Users/0x3333/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "97be7950-053a-47ac-beb9-6cf92857e148",
    "pvtShowDevDebugOutput": "raw"
}
0000000004+00002: Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/0x3333/Code/Home/g6/build/gateway.elf
[... 262 lines cut ...]
0000001873+00008: -> =breakpoint-modified,bkpt={number="2",type="breakpoint",disp="del",enabled="y",addr="0x10000428",func="main",file="/Users/0x3333/Code/Home/g6/src/main.c",fullname="/Users/0x3333/Code/Home/g6/src/main.c",line="22",thread-groups=["i1"],times="1",original-location="-function main"}
0000001874+00001: -> ~"\n"
0000001874+00000:
0000001874+00000: -> ~"Temporary breakpoint 2, main () at /Users/0x3333/Code/Home/g6/src/main.c:22\n"
0000001874+00000: Temporary breakpoint 2, main () at /Users/0x3333/Code/Home/g6/src/main.c:22
0000001875+00001: -> &"warning: Source file is more recent than executable.\n"
0000001875+00000: warning: Source file is more recent than executable.
0000001875+00000: -> ~"22\t{\n"
0000001875+00000: 22	{
0000001875+00000: -> *stopped,reason="breakpoint-hit",disp="del",bkptno="2",frame={addr="0x10000428",func="main",args=[],file="/Users/0x3333/Code/Home/g6/src/main.c",fullname="/Users/0x3333/Code/Home/g6/src/main.c",line="22",arch="armv6s-m"},thread-id="1",stopped-threads="all"
0000001875+00000: mi2.status = stopped
0000001875+00000: -> =breakpoint-deleted,id="2"
0000001876+00001: 21-thread-list-ids
0000001876+00000: -> 21^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
0000001877+00001: 22-thread-info 1
0000001877+00000: -> 22^done,threads=[{id="1",target-id="Thread 1",frame={level="0",addr="0x10000428",func="main",args=[],file="/Users/0x3333/Code/Home/g6/src/main.c",fullname="/Users/0x3333/Code/Home/g6/src/main.c",line="22",arch="armv6s-m"},state="stopped"}]
0000001878+00001: 23-stack-list-frames --thread 1 0 19
0000001878+00000: -> 23^done,stack=[frame={level="0",addr="0x10000428",func="main",file="/Users/0x3333/Code/Home/g6/src/main.c",fullname="/Users/0x3333/Code/Home/g6/src/main.c",line="22",arch="armv6s-m"}]
0000002284+00406: 24-stack-info-frame --thread 1 --frame 0
0000002284+00000: -> 24^done,frame={level="0",addr="0x10000428",func="main",file="/Users/0x3333/Code/Home/g6/src/main.c",fullname="/Users/0x3333/Code/Home/g6/src/main.c",line="22",arch="armv6s-m"}
0000002288+00004: 25-stack-list-variables --thread 1 --frame 0 --simple-values
0000002288+00000: -> 25^done,variables=[{name="wait",type="uint_fast8_t",value="<optimized out>"}]
0000002289+00001: 26-var-create --thread 1 --frame 0 var_wait_4096 @ "wait"
0000002289+00000: -> 26^done,name="var_wait_4096",numchild="0",value="<optimized out>",type="uint_fast8_t",has_more="0"
```

dragonmux, 2024-11-01:
Hi there, unfortunately the Cortex-Debug output isn't particularly useful to us in this case as it doesn't give a why, from BMD's side of things (BMD is unable to express to GDB where and why Flash programming went wrong).

What would be useful to help us help you is: what version of the firmware are you running on your Black Pill? and if you run up BMDA as `blackmagic -v 5` and connect to that to do operations, what does that say on the console BMDA is run on?

That latter part will connect to your Black Pill probe using the BMD remote protocol, and then have BMDA drive everything. It will be slower, but BMDA has complete diagnostics available and is the easiest way to access the log output generated for what's going on with a target.

0x3333, 2024-11-01:
Sounds fair, but I don't know how to reproduce what cortex-debug is doing, I can load using gdb but that's all. I'll close as I can't help myself in this. I appreciate your efforts.

0x3333, 2024-11-01:
So, I read GDB MI documentation, and been able to reproduce the way you need it the same commands cortex-debug did:

```
17:01:05 /Users/0x3333/Code/blackmagic/build $ ./blackmagic -v 5
Black Magic Debug App v1.10.0-1429-g8027f815
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Using 1d50:6018 316F378F3132 Black Magic Debug
 Black Magic Probe (BlackPill-F411CE) v1.10.0-1429-g8027f815
Setting V6ONLY to off for dual stack listening.
Listening on TCP port: 2000
Got connection
Speed set to 3.000MHz for SWD
Switching from JTAG do dormant
Switching out of dormant state into SWD
Handling SWD multi-drop, TARGETID 0x01002927
DP DPIDR 0x0bc12477 (v2 MINDP rev0) designer 0x43b partno 0xbc
TARGETID 0x01002927 designer 0x913 partno 0x1002
AP   0: IDR=04770031 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var3 rev0)
Halt via DHCSR(00030003): success after 3ms
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4c0 (PIDR = 0x04000bb4c0)
0 0x0e000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
cortexm_probe: Examining Part ID 0x1002, AP Part ID: 0x04c0
Calling rp2040_probe
1 0x0e0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a DEVTYPE = 0x00 ARCHID = 0x0000)
2 0x0e0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b DEVTYPE = 0x00 ARCHID = 0x0000)
ROM Table: END
DP DPIDR 0x0bc12477 (v2 MINDP rev0) designer 0x43b partno 0xbc
TARGETID 0x01002927 designer 0x913 partno 0x1002
AP   0: IDR=04770031 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var3 rev0)
Halt via DHCSR(00030003): success after 2ms
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4c0 (PIDR = 0x04000bb4c0)
0 0x0e000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
cortexm_probe: Examining Part ID 0x1002, AP Part ID: 0x04c0
Calling rp2040_probe
1 0x0e0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a DEVTYPE = 0x00 ARCHID = 0x0000)
2 0x0e0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b DEVTYPE = 0x00 ARCHID = 0x0000)
ROM Table: END
DP DPIDR 0x10212927 (v2 MINDP rev1) designer 0x913 partno 0x2
TARGETID 0x01002927 designer 0x913 partno 0x1002
RP2040 Flash controller not in POR state, reconfiguring
Flash device ID: ef 40 15
8 byte SFDP read at 0x0:
	53 46 44 50 05 01 00 ff
8 byte SFDP read at 0x8:
	00 05 01 10 80 00 00 ff
64 byte SFDP read at 0x80:
	e5 20 f9 ff ff ff ff 00
	44 eb 08 6b 08 3b 42 bb
	fe ff ff ff ff ff 00 00
	ff ff 40 eb 0c 20 0f 52
	10 d8 00 00 36 02 a6 00
	82 ea 14 b3 e9 63 76 33
	7a 75 7a 75 f7 a2 d5 5c
	19 f7 4d ff e9 30 f8 80
Flash size: 2MiB
rp_flash_prepare
Write failed at 10000c70
rp_flash_resume
```

Here is the GDB output:

```
arm-none-eabi-gdb build/gateway.elf --interpreter mi3
=thread-group-added,id="i1"
~"GNU gdb (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 14.2.90.20240526-git\n"
[... 55 lines cut ...]
^done
(gdb)
-interpreter-exec console "attach 1"
~"Attaching to program: /Users/0x3333/Code/Home/g6/build/gateway.elf, Remote target\n"
=thread-group-started,id="i1",pid="1"
=thread-created,id="1",group-id="i1"
*running,thread-id="1"
^done
~"0x10008e20 in prvIdleTask (pvParameters=<optimized out>) at /Users/0x3333/Code/Home/g6/deps/FreeRTOS-Kernel/tasks.c:5809\n"
~"5809\t            if( listCURRENT_LIST_LENGTH( &( pxReadyTasksLists[ tskIDLE_PRIORITY ] ) ) > ( UBaseType_t ) configNUMBER_OF_CORES )\n"
*stopped,frame={addr="0x10008e20",func="prvIdleTask",args=[{name="pvParameters",value="<optimized out>"}],file="/Users/0x3333/Code/Home/g6/deps/FreeRTOS-Kernel/tasks.c",fullname="/Users/0x3333/Code/Home/g6/deps/FreeRTOS-Kernel/tasks.c",line="5809",arch="armv6s-m"},thread-id="1",stopped-threads="all"
-interpreter-exec console "set mem inaccessible-by-default off"
=cmd-param-changed,param="mem inaccessible-by-default",value="off"
^done
(gdb)
-target-download
+download,{section=".boot2",section-size="256",total-size="1037794"}
+download,{section=".boot2",section-sent="256",section-size="256",total-sent="256",total-size="1037794"}
+download,{section=".text",section-size="44224",total-size="1037794"}
^error,msg="Error writing data to flash"
(gdb)
```

What I could understand is:
The first flash fails, then the next time, the firmware is corrupted(The application doesn't start even after a restart). With the firmware corrupted or empty, I can run target-download and everything works.

dragonmux, 2024-11-01:
How interesting! Yes, this is something unspecified going wrong at +0xc70 into Flash on latest `main`. Likely that means that either the second core isn't halted during the operation, causing the Flash stub to have a problem, or there's some other state that happens that gets it unhappy.

The error reporting for the RP2040 Flash process is not amazing in part because the main body of work that's done takes place directly on the target itself via a stub that's loaded on during rp_flash_prepare(). We'll have a think on how we can improve that and get the stub and the control code to have a bit more information on what exactly is going wrong to improve diagnostics.

0x3333, 2024-11-01:
The error happens when my code is running which in fact uses the second core. Anything I can do to help?

dragonmux, 2024-11-01:
Thank you for the extra information. That confirms our suspicion that there is some re-architecture of BMD required to allow us to express that multiple cores are ganged together, and that when we halt one of them we need to halt all of them for things like entering Flash mode.

Given that the project dynamically discovers the part and all cores, this is going to be a bit tricky which is why we'd been trying to not have to do the change required, but you've given us a clear example of a configuration of a part where the only way to get the right behaviour and a successful Flash programming is to halt all cores.

If you wanted to, you could look at how to add a member to the target structure for describing core cross-linking with a list of other targets that participate in being the same physical device that all need halting together; however we completely understand if you'd prefer to leave it with us and let us dig in.

Note that any solution to this needs to factor in that a scan chain may have multiple physical devices on it each providing targets for debugging, possibly even of the same device (eg, having multiple LPC43xx's on the same JTAG scan chain), and so the cross-linking cannot just assume that all targets, or even all targets of the same part name, are part of the same SoC/MCU.

0x3333, 2024-11-02:
> If you wanted to, you could look at how to add a member to the target structure for describing core cross-linking with a list of other targets that participate in being the same physical device that [...]

I'm affraid that this is outside my area of knowledge. C is no problem, but the internals of the debugger that isn't.

If you need any extra information, just let me know. Thank you again for your efforts.