blackmagic-debug/blackmagic #2001: FR: Make full BMDA for Windows inside Github actions

positron96, 2024-11-12:
At the moment Windows version of BMDA built by Github actions contains FTDI connection option only. The rest of the options, libusb-based and hidapi-based, are not compiled in. Would be nice to have an automated windows build that produces binaries with all of them.
I've made a proof of concept of changes to workflow in my fork and can make a PR.

dragonmux, 2024-11-12:
👋🏼 The `build-and-upload` CI workflow builds BMDA for Windows using MSYS2's UCRT64 environment and statically builds libusb and hidapi. Full BMDA builds are not possible without including all 3 dependencies, but you won't see separate DLLs for libusb and hidapi because the resulting binaries are statically linked into the BMDA executable.

https://github.com/blackmagic-debug/blackmagic/actions/runs/11782970941/job/32819176218 is an example - if you open the `Build full BMDA` item and scroll down to the summary section of Meson's setup step, you'll be able to see in the `Subprojects` section both `hidapi` and `libusb` being built  statically.

The Meson build system is intentionally incapable of building a BMDA that is for the BMP remote protocol only, so necessarily includes all BMDA backends which is why those dependencies are required.

positron96, 2024-11-12:
Oh, thanks, I didn't know that! Last time I checked BMDA build, about half a year ago, it couldn't start without some DLLs. 
But now I see that ftd2xx.dll is bundled into the artifact, is it linked dynamically?

dragonmux, 2024-11-12:
Aye, correct - it's the FTDI provided Windows counterpart to libftdi so users don't have to mess with Zadig to rebind drivers to use their FTDI-based adaptors, making full BMDA for Windows just two files for a fully working setup.

positron96, 2024-11-13:
Ok! Not an issue then

closed 2024-11-13 by positron96

positron96, 2024-11-14:
A somewhat related question though. Is it expected to work with STLink v2 clone? It looks like it hangs on the 2nd transaction:
```
D:\TMP>blackmagic-bmda.exe -v 255
Black Magic Debug App 3576cdf
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Using 0483:3748 1E3609013212364D434B4E00 STMicroelectronics
 STM32 STLink ---
 request: f1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
response: 27 47 83 04 48 37
ST-Link firmware version: V2J29S7
Trying ST-Link reset
 request: f1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```